OpenSatKit/OpenSatKit #2: Incomplete Documentation

smorad, 2017-11-09:
It looks like a lot of the sections in the user guide are "tbd". Have there been any updates to this doc that hasn't been synced to github? OpenSatKit would be really useful to us, but documentation for this and cFS is quite sparse.

dmccomas, 2017-11-15:
Hi Steven,

I’m currently updating the user’s guide with a hope of this weekend for a releasable update.  Still won’t be complete but better than it is now!

Glad you’re interested in the kit.  What are you using it for?

-          Dave

smorad, 2017-11-16:
We're looking at software options for SWIMSat: 

https://digitalcommons.usu.edu/cgi/viewcontent.cgi?referer=https://www.google.com/&httpsredir=1&article=3442&context=smallsat

dmccomas, 2017-11-24:
Hi Steven,

I’ve attached an updated User’s Guide include in OpenSatKit release 1.0.  I’ve populated the application management section. Still more work to do…

My goals with OpenSatKit are to create a freely available cFS-based educational tool.  You may be interested in another effort called the NASA Operational Simulation for Small Satellites (NOS3). They’re also using COMSOS & cFS but they’re goal is to support the spacecraft development lifecycle.  It’s more powerful but also more complicated. Eventually I’d like to have the two efforts compatible. For started we could share the same COSMOS database. Some info on NOS3: http://www.stf1.com/simulation.php,  https://www.nasa.gov/centers/ivv/jstar/nos3.html

There’s also a cFS workshop on Monday 12/4. More info at http://flightsoftware.jhuapl.edu/

-          Dave

https://digitalcommons.usu.edu/cgi/viewcontent.cgi?referer=https://www.google.com/&httpsredir=1&article=3442&context=smallsat