OpenSatKit/OpenSatKit #3: COSMOS doesn't start after the initial installation launch

dmccomas, 2017-11-23:
The installation appears to work fine: COSMOS is launched and the cFS can be started. However, after COSMOS is shutdown and you try to restart COSMOS you get an error message stating ruby isn't installed.  Ubuntu recommends installing ruby. Do not do this.  If you did and try to launch COSMOS, you'll get a ruby require.  If you installed ruby you can uninstall it by running "sudo apt-get purge ruby"

I have found that rerunning the ruby portion of the installation script fixes the issue. 
1. Goto https://github.com/OpenSatKit/OpenSatKit/blob/vendor/install.sh
2. Copy the portion of the script between _announce "Installing ruby 2.2.6 using rbenv"_ and _announce "Installing COSMOS gem"_
3. Paste the script excerpt into a terminal window to execute it.

OSK 1.1 will be upgraded to use COSMOS 4.x so I'm hoping this will resolve any ruby /COSMOS issues.

kihwankim, 2020-10-21:
I also wasted a lot of time because of this problem. But if you install rbenv at first, it doesn't matter -> someone will install this project in your local computer please install rbenv at first.

dmccomas, 2020-10-21:
Did you have a problem with the current installation?  
Are you saying that if rbenv is installed before the COSMOS installation it works?  I was relying on the COSMOS installation to install rbenv.

Thanks - Dave

kihwankim, 2020-10-23:
I had a problem when I installed this about 2 months ago cuz of ruby version problem. but after I installed rbenv first, it worked out successfully in Ubuntu 18.04