OpenSatKit/OpenSatKit #6: Can't get OpenSatKit to work with many flavors of Ubuntu

tme31277, 2018-06-21:
I have had issues with the installation script that you post in the instructions page: 
bash <(\wget -qO- https://raw.githubusercontent.com/OpenSatKit/OpenSatKit/vendor/install.sh)

With 17.10.1 & 18.04 Ubuntu/64-bit, both of these will not install ruby correctly, and even if I try and install the ruby 2.2.6 version from the install.sh script manually, I get various errors that prevent cosmos from starting.  

With 16.04.04 & 14.04 Ubuntu/64-bit, everything installs, but JSON-RPC always errors out when launching the Packet Viewer.

I have had success with 16.04 LTS Ubuntu/32-bit, but unfortunately, one day the hard drive failed and I am reinstalling that system now, so I don't know if it will work the next time.

Is there a troubleshooting section in the user guide?  I couldn't find one.

dmccomas, 2018-06-25:
I just uploaded a new installation script last night that should work with all versions of Ubuntu thanks to my intern. The Ubuntu bash is changing and it impacts the ruby installation. My intern was working with the primary author of COSMOS and eventually made it work.  After the initial installation you still must close the first terminal window and then open a new one to keep the ruby paths intact.

As far as a trouble shooting guide I’ve logged a couple of FAQs in the user’s guide and that’s it.  OSK grew out of the need for people to learn about the cFS and have an operational system.  It’s only me and my goal is to find the bandwidth to manage OSK as an open source project that accepts contributions.

-        Dave

tme31277, 2018-06-28:
I found the source of the problem of the JSON issues.  It was because I didn’t have the no_proxy environment variable set up.  It is there in the documentation on the COSMOS website:

The http_proxy environment variable can cause problems. Make sure you also have a no_proxy variable for localhost, something like no_proxy=”127.0.0.1, localhost”.

From the website below:
http://cosmosrb.com/docs/installation/

closed 2018-06-28 by tme31277