OpenSatKit/OpenSatKit #7: cFS installation on Raspberry Pi

josecarlosmd, 2018-06-22:
Hello everyone,

I'm going through some problems with the installation on Raspberry Pi 3 B+. I remove every reference to x86 architecture and so on, but when I try 'make' I get the following error:

Makefile:34: /pc-linux/make/compiler-opts.mak: No such file or directory
Makefile:38: /posix/osal.mak: No such file or directory
Makefile:59: /make/make-rules.mak: No such file or directory
make[2]: *** No rule to make target '/make/make-rules.mak'.  Stop.
make[2]: Leaving directory '/home/pi/CFE/cFE-6.5.0-OSS-release/build/cpu1/cfe/osal'
Makefile:3: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/pi/CFE/cFE-6.5.0-OSS-release/build/cpu1/cfe'
../cfs.mak:145: recipe for target 'cfs_build_core' failed
make: *** [cfs_build_core] Error 2

Any ideas? Thank you in advance

Immortalin, 2018-06-22:
Look up Perihelion OS, prebuilt raspbian with cFS integration

josecarlosmd, 2018-06-23:
Thank you I will! I'm going to give it a try, as firstly I want to check out cFS, however, I would like to solve the issue, as eventually I'll have to install it over a RTOS. Any suggestion? Thank you again

Immortalin, 2018-06-24:
@josecarlosmd I haven't touched it in quite a while but if you unpacked the Perihelion iso, there should be a copy of OpenSatKit with the correct makefile configs

dmccomas, 2018-06-27:
Did you run “setvars.sh” ?

$  .  ./setvars.sh

-        Dave

josecarlosmd, 2018-06-27:
Dear Davis,

Yes, I ran it, with no errors, apparently. Thanks.

Immortalin, 2018-06-27:
@josecarlosmd direct link for your convenience: https://github.com/Immortalin/Perihelion_OS/releases

Immortalin, 2018-06-27:
The raspbian version is a bit outdated but the cFE should work perfectly fine

aogrcs, 2018-11-14:
@Immortalin 
Hi, thanks for your work. I compare between your cFE code and the original, the differences are comments on some statements about m32, is this right? Thanks

Immortalin, 2018-11-14:
I believe those were the main changes yes, lemme know if you face any issues building it

Immortalin, 2018-11-14:
Oh, and ensure you use a good memory card. CFe is rather write intensive (logging maybe?). Burnt through a few cards, check for memory leaks too.

evanfitzgerald, 2020-06-17:
Consider this for installation on the Raspberry Pi : https://alanstechnotes.blogspot.com/2018/11/how-to-compile-core-flight-system-on.html

dmccomas, 2020-07-01:
Thanks Evan!  Expanding the range of deployments is definitely on the
strategic path.  - Dave