OpenSatKit/OpenSatKit #8: Unit_Test up to date ?

ybonth, 2018-06-26:
Hello everyone,

I am trying to run the Unit test of different apps, (sc  and UTF mainly).
I was wandering if the files concerning Unit test where up to date or if I missed something obvious to do before trying to run it. 
For the moment I only run setvars.sh and then try to make the test (with Sc for exemple). I tried other apps and at some point I saw that the unit test framework didn't match the 6.5.0 version of the cFE. 
For exemple, in the previous version of cFE in cfe_es.h CFE_ES_DeviceDrivert was defined (type used in utf_cfe_es_api).

Am I missing something important before running the unit test ? Is it up to date ? 
If you have any question about my demand feel free to ask.

Sorry for my engish and thank you for your time !

dmccomas, 2018-07-02:
I did not verify that the application unit tests would build and run when I created the kit. I can look into it.  

However I don't want to put too much effort into cFE 6.5 because cFE 6.6 has made some changes to the unit tests.   Unfortunately  cFE 6.6 has not finished going through the open source release process. At the moment it is only available for US government only.

ybonth, 2018-07-02:
Thank you for your answer, let me know if you find something to build and run unit tests. In case it doesn't work, I'll wait for the 6.6 update. 

Thank you for your help !

dmccomas, 2018-10-30:
Sorry for the delayed response.  Unfortunately enhancing the unit test within the cFE 6.5 context has become a lower priority.  The recent focus has been in improving the creation and integration of apps. The open source release of cFE 6.6 is still hung up in the release process. Once cFE 6.6 is released and integrated into OSK I can look into improving the unit test support.