OpenSatKit/OpenSatKit #9: Health & Safety App Crashes with Ubuntu versions greater than 16.04

dmccomas, 2018-07-01:
I haven't had time to investigate the cause so the easier work around is to comment out the HS app in the executive services startup script: OpenSatKit-master/cfs/build/exe/cpu1/cf/cfe_es_startup.scr. Note that ES stops processing the startup script after it reaches the first comment line so you must copy the HS line to the end of the file with the other commented lines.

Also note if you rebuild the FSW then  OpenSatKit-master/cfs/osk_defs/cpu1_cfe_es_startup.scr gets copied to the "../cf/" directory with the "cpu1_" removed (see users guide). Therefore HS also needs to be commented out in cpu1_cfe_es_startup.scr as well.

jmdagostino, 2018-07-02:
Be careful when commenting out HS in the cfe_es_startup.scr files.  If I remember correctly, when ES encounters a commented-out line in the startup script it stops processing the rest of the script, so no apps after the commented-out line will be loaded.
It was a long time ago that encountered this, but I haven't heard about any modifications to script processing recently.
The approach I took was to move all commented-out apps to the end of the list.

dmccomas, 2018-07-02:
Thanks. I’m aware. I thought I put this warning in the reply. I can check later

jmdagostino, 2018-07-02:
...and I just re-read your original message, and yes, you already said that.
I'll just claim that I was repeating the information for emphasis and go sit in the corner now...

dmccomas, 2018-07-02:
I appreciate the intention!