OpenSatKit/OpenSatKit #10: 42 Simulation does not run multiple times

dmccomas, 2018-07-12:
If the 42 simulation is run to completion it won't run again without restarting the VM.  If the 42 simulation is interrupted then the simulation can be restarted without restarting the VM.

derricku, 2019-10-23:
Confirmed having the same issue. 
When forced to quit it shows an "Unexpected Text" log with repeating "Input sudo password" and "Incorrect password entered" messages. 
42 seems to expect the sudo password like usual but fails to open a terminal input to do so.

dmccomas, 2019-11-19:
Version 2.0 changed the 42-cFS relationship.  42 is now started with a very long simulation time with the goal of not letting 42 terminate on its own. 42 should only be terminated with a disconnect command and there aren't any issues with restarting 42 after it is disconnected.

closed 2019-11-19 by dmccomas