OpenSatKit/OpenSatKit #13: Connection refused

BottomBanana, 2019-05-03:
Hello!
I'm having trouble getting cosmos and cfs telemetry to work. 
I click on cFS starter kit as per the instructions then all the correct windows pop up. 
The time counter is on magenta 0. I click the Start cFS button and get a terminal window asking for my password. I enter it and all seems well but the time counter is still on magenta 0. Clicking Enable Telemetry a couple of times results in an error message: 'Errno::ECONNREFUSED : Connection refused - Connection refused'.
I get a similar error when trying to enable telemetry through PiSat control.
I'm running Ubuntu MATE 18.04.2 on a Raspberry Pi 3 B+
Kernel Linux 4.15.0-1032-raspi armv7l 

Many thanks
Stephen

dmccomas, 2019-05-04:
Hi Stephen,

The enable telemetry error message indicates that COSMOS and the cFS (running in the terminal window) have not connected.  Do you see an “enable telemetry” message in the terminal window when you click the button? I’m guessing no.

When you first start the kit from the COMSOS launcher one of the windows that pops up is the COSMOS cmd-tlm server.   I think it defaults to the connection tab. I don’t remember exactly what it’s called and I’m away from my OSK computer.
What’s the status of the connections? “CFS_INT” is the important one. Or something close to that name.

COSMOS and the cFS communicate over localhost 127.0.0.1.

Don’t mess with the PiSat connection. That’s for a special wireless configuration.  One of my goals in the next few months is to improve documentation!!

I’ll be back Sunday evening and can help more then.

-        Dave

BottomBanana, 2019-05-05:
Hi Dave
Thank you for a swift reply. Please see the screen dump below.
The terminal window remains on the ruby Launcher command and the connections are all green.

Many thanks for your help

[image]

dmccomas, 2019-05-09:
I know you sent a reply but it seems to have disappeared.  They're "upgrading" our work email and things have been a little scrambled this week. Can you send it to my cc'd email.

Thanks - Dave

________________________________
From: Mccomas, David C. (GSFC-5800)
Sent: Saturday, May 4, 2019 8:40 AM
To: OpenSatKit/OpenSatKit
Subject: RE: [OpenSatKit/OpenSatKit] Connection refused (#13)

Hi Stephen,

The enable telemetry error message indicates that COSMOS and the cFS (running in the terminal window) have not connected.  Do you see an “enable telemetry” message in the terminal window when you click the button? I’m guessing no.

When you first start the kit from the COMSOS launcher one of the windows that pops up is the COSMOS cmd-tlm server.   I think it defaults to the connection tab. I don’t remember exactly what it’s called and I’m away from my OSK computer.

What’s the status of the connections? “CFS_INT” is the important one. Or something close to that name.

COSMOS and the cFS communicate over localhost 127.0.0.1.

Don’t mess with the PiSat connection. That’s for a special wireless configuration.  One of my goals in the next few months is to improve documentation!!

I’ll be back Sunday evening and can help more then.

-        Dave

BottomBanana, 2019-05-10:
Dear Dave

I hope this finds you well. My (rather short) reply is visible on the
github site via this link.
https://github.com/OpenSatKit/OpenSatKit/issues/13
I have attached a screen grab of the windows. As you can see the terminal
window simply shows the ruby Launcher command and the telemetry windows I
think shows four good connections.
If necessary we could arrange a remote desktop session for you to have a
poke around. I'm very grateful that you take the time to tackle these
issues.

Kind regards

Stephen

BottomBanana, 2019-05-14:
Hi Dave

Did you get my  last message with the screen grab? The conversation is also
visible on the github messages page.
https://github.com/OpenSatKit/OpenSatKit/issues/13

Regards
Stephen

dmccomas, 2019-05-14:
After you entered the OSK password did the cFS terminal window stay open?  After you enter your password you should see a bunch of start up messages ending with something similar to the picture below. They won't be exact but it should be close.

- Dave

https://github.com/OpenSatKit/OpenSatKit/issues/13

dmccomas, 2019-05-15:
Hi Stephen,

My email troubles should be behind me...  Here’s the cFS startup screen. This is my development platform so it may be a little different.  Note the I42 connection failure is for a 42 simulator interface app and does impact the cFS-to-COSMOS communication.

-        Dave

[cid:image001.jpg@01D50B0E.299F4890]

-        Dave

https://github.com/OpenSatKit/OpenSatKit/issues/13

BottomBanana, 2019-05-15:
Hi David

I cannot see the image you sent. Perhaps we should just use the message
tools on the github message page via a browser.

regards

Stephen

dmccomas, 2019-05-21:
When you click start you should see a prompt for the password. After you enter the password abunch of startup messages will appear and the last few should look similar to this image.
[image]

BottomBanana, 2019-05-21:
After entering the password that Terminal window closes. I see nothing that resembles your image.

dmccomas, 2019-06-04:
Check to see if you have a file named core-cpu1 in   [opensatkit-master]/cfs/build/exe/cpu1.  If not then the cFS was not built during installation. Section 4.1.1 of the User's Guide ([opensatkit-master]/docs/OSK_UsersGuide) has instructions for building the cFS. 

If core-cpu1 exists then create a copy of [opensatkit-master]/cfs/build/exe/cpu1/cfe_es_startup.scr. In the original named version delete all of the entries or you can comment them out with a "!" at the start of each line. The bottom line is you don't want anything loaded.  Then try to start the cFS. Thsi will tell us if the core is having a problem or an app or library is crashing during startup.

bapichulo, 2019-06-25:
[image]

Does anyone have a solution to the issue in this screenshot?

pavja2, 2020-01-28:
> Check to see if you have a file named core-cpu1 in [opensatkit-master]/cfs/build/exe/cpu1. If not then the cFS was not built during installation. Section 4.1.1 of the User's Guide [...]

I know this is an old issue but I can confirm that I was having the same problem (enter sudo password and then terminal window silently closes and System Time never starts counting up). Following the suggestion to manually build cfs (now section 5.1.1 of the User's Guide) fixed it for me.

PaulaBosca, 2020-05-07:
Hello, 
I have a similar problem. I am running Ubuntu 16.04.6 on a virtual machine. I managed to download OpenSatKit and after using bundle install I managed to get ruby Launcher to work. It successfully opens COSMOS and all the associated windows. However, when I click 'start cFS', a terminal opens prompting me to input my password. Once I do, the terminal disappears, the COSMOS windows are greyed out for a couple seconds (the timer counter doesn't increase) and then nothing happens. 
I tried following the manual cfs build in section 5.1.1 in the User's guide but I get a fatal error that there is not jsmn.h file. 
I'm trying to learn cFS for a CubeSat mission at my University and would greatly appreciate any help! Thank you!

dmccomas, 2020-05-12:
Did you download it recently? There was a period of time when I had the wrong cfe_es_startup.scr file in the release. Thsi file defines whcih apps are loaded during startup. This file occurs in two places. 
1.  .../cfs/osk_defs/cpu1_cfe_es_startup.scr is the main definition. When the cFS is built the 'cpu1_' prefix is stripped off and the file is copied to  
2. .../cfs/build/exe/cpu1/cf/es_startup.scr

The health & safety (HS) app should not be loaded because it crashes on some platforms. Your startup script should have HS commented out using or he line deleted. Note that the first '!' encountered comments out the remainder of the file.

PaulaBosca, 2020-05-15:
Oddly enough, the HS app doesn't load since it's correctly commented out in cfs/osk_defs/cpu1_ed_startup.scr 
However, when my system builds it doesn't create the build/exe folder. the cfs/build folder contains the following directories: CMakeFiles, cpu1, doc, editor_inc, inc, tools, and two text files (CMakeCache.txt and missions_vars.cache). The cfs/build/cpu1 does not have the cf files or the like. Any idea what could be happening with my build? I've tried reinstalling within the last couple days with the same problems.

dmccomas, 2020-05-21:
I don't know what's going on.  I have time for a meeting tomorrow (5/21)
3:30pm EST or later.  I could setup a microsoft teams meeting where you
could show me what's going on.

PaulaBosca, 2020-05-21:
That sounds wonderful! Thank you so much! I'm also free anytime after 3:30 today

dmccomas, 2020-05-21:
Got busy at work.    Let's do  4:30pm EST.

________________________________________________________________________________

Join Microsoft Teams Meeting
<https://teams.microsoft.com/l/meetup-join/19%3ameeting_ODE4MzU3NzktNjYzZS00MjE2LThiMGQtZjZiMzg1MjdkNWZk%40thread.v2/0?context=%7b%22Tid%22%3a%227005d458-45be-48ae-8140-d43da96dd17b%22%2c%22Oid%22%3a%22d3059c77-7a67-4cd2-ba80-599c704abdb3%22%7d>

Learn more about Teams <https://aka.ms/JoinTeamsMeeting> | Meeting options
<https://teams.microsoft.com/meetingOptions/?organizerId=d3059c77-7a67-4cd2-ba80-599c704abdb3&tenantId=7005d458-45be-48ae-8140-d43da96dd17b&threadId=19_meeting_ODE4MzU3NzktNjYzZS00MjE2LThiMGQtZjZiMzg1MjdkNWZk@thread.v2&messageId=0&language=en-US>

Help <https://nasa.sharepoint.com/sites/TeamsSupport>   |  Legal
<https://nasa.sharepoint.com/sites/TeamsSupport/SitePages/Teams-Disclaimer.aspx>

dmccomas, 2020-06-13:
Sorry for the delay & or teh code that's cuasing your issue.  The window closes because the cFS isn't being built and there's no executable.   There was a short period of time when I had commited some intermittent C++ app code that wouldn't  build.  The current repo code works.   If you don't want to download the latest code code you can you can remove the demo apps from the cmake target list file.  The C demo app had issues at the same time the C++ demo app had issues. 

To remove the demo apps from being built edit  OpenSatKit-master/cfs/osk_defs/targets.cmake

Remove the demo apps from TGT1_APPLIST
FROM:
SET(TGT1_APPLIST cfs_lib expat_lib osk_c_fw osk_cpp_fw cs ds fm hs kit_ci kit_sch kit_to lc md mm sc hk tftp i42 f42 isim hc hsim osk_c_demo osk_cpp_demo)

TO:
SET(TGT1_APPLIST cfs_lib expat_lib osk_c_fw osk_cpp_fw cs ds fm hs kit_ci kit_sch kit_to lc md mm sc hk tftp i42 f42 isim hc hsim)

Hope this fixes it.

Natpanzarini, 2020-06-15:
Hi, thanks for the response!

I have both reinstalled and removed the demo apps from TGT1_APPLIST as instructed, but cFS still does not build when I run make. It looks like there are issues when it tries to build osk_cpp_fw, as it always throws errors at this point in the process.

I have attached my errors again. Do you have any other suggestions for fixing the cFS build issue?

Thanks!
[cfs_errors.txt](https://github.com/OpenSatKit/OpenSatKit/files/4781956/cfs_errors.txt)

dmccomas, 2020-06-24:
Sorry for the delay.  I'm not sure why the nullptr is not resolved but I
think the easiest solution to get you started using OSK is to remove the
C++ library and C++ demo app from the build target list and from the
cfe_es_startup.scr.  The C++ app framework and demo app are recent
additions and the rest of the kit does not depend on them.  I'll continue
to look into the C++ code issue.

In the cfs/osk_defs directory edit

1. targets.cmake
Remove  'osk_cpp_fw'  and 'osk_cpp_demo' from the following line

SET(TGT1_APPLIST cfs_lib expat_lib osk_c_fw osk_cpp_fw cs ds fm hs kit_ci
kit_sch kit_to lc md mm sc hk tftp i42 f42 isim hc hsim osk_c_demo
osk_cpp_demo)

2. cpu1_cfe_es_startup-dev.scr

Remove the following line or move it to the end of the file. The C++ demo
app is not loaded by default.  Note that you can't simply comment out the
line with '!' because Executive Services stops loading apps when it
encounters the first '!'

CFE_LIB, /cf/osk_cpp_fw.so,   OSK_CPP_FW_LibInit,  OSK_CPP_FW,     0,
16384, 0x0, 0;

Hope this helps. .

pmepablo, 2020-08-07:
Thanks dmccomas, editing those files did the trick for me.  I was also getting the nullptr prior to updating those two files.