OpenSatKit/OpenSatKit #15: Cannot connect to server

bapichulo, 2019-06-25:
After completing the installation, I have been having issues launching cosmos using 'ruby Launcher'. I have tried typing 'bundle install' after I read the issues in this forum page, but after typing in ruby launcher, the terminal then says "Launcher: cannot connect to X server".

Does anyone have a solution to this? If there is more information that I need to post on my side, please let me know.

dmccomas, 2019-07-17:
Which version of Ubuntu are you using?

rsmadja, 2020-06-16:
Hello I just downloaded OpenSatKit, end after a few issues with the instalation on ubuntu 20.04, I scaled down to the 18.04 version. Now the installation runs completely but I have this same issue where after I call : '~/cosmos/$ ruby Launcher', The terminal says 'Launcher: cannot connect to X server'

If anyone could help me that would great !
Thank you in advance !

SadiqSarwar, 2021-02-22:
I am also getting this error. I am currently using Ubuntu 18.04 LTS terminal on my windows 10 OS. Has any solution to this error message been identified?

SadiqSarwar, 2021-02-24:
I realized I did not have any X Server installed on my computer. I had to install an X Server like VcXsrv, which solved the problem. Then the command 'ruby Launcher' actually opened the GUI for me.

I followed this stackoverflow answer as a solution. But instead of gedit, I ran the 'ruby Launcher' command.
[https://stackoverflow.com/questions/60284542/wsl-gedit-unable-to-init-server-could-not-connect-connection-refused](url)