OpenSatKit/OpenSatKit #16: TO send CADU frames

Amr-Navy, 2019-07-12:
how to configure TO send CADU Frames ??

dmccomas, 2019-07-17:
NASA Goddard was not able to open source their CI and TO apps used for missions. NASA's Johnson Space Flight Center created open source versions of CI and TO for technology efforts. CI & TO depend on IO_LIB.  I have not used them and don't know how well they've been tested, but I see CADU references in IO_LIB.

https://github.com/nasa/CFS_CI
https://github.com/nasa/CFS_TO
https://github.com/nasa/CFS_IO_LIB

We have a cFS mailing list. I can post the question or if you'd like to join let me know and I'll add you to the list so you can post.

Amr-Navy, 2019-07-18:
I know about these but still can't use IO_LIB and their documents not enough for me. So, please post the question and add me to the mailing list.
Thanks

dmccomas, 2019-08-21:
Sorry for the delay.  I don’t have your email address to add to the list since this conversation has been through the git issues. You can send your address directly to dmccomas63@gmail.com<mailto:dmccomas63@gmail.com> if you’d prefer.

I’m going on vacation tomorrow until 8/27 but will have occasional email access. Once you’re on the list it’s probably best if you post your specific situation/question.

Thanks - Dave

Amr-Navy, 2019-10-01:
Dear Sir
Thanks for your support i have succeeded  to make to_lab send CADU frames.
Now i am working on a project to Encrypt and Authenticate these CADU frames
then verified and decrypted it in ground using YAMCS as my ground Station
i want to know your opinion in this project does it worth and will be
useful or you can recommended me someone who is interested  in this project.

Thanks in advance
Best Regards

closed 2021-04-30 by Amr-Navy