OpenSatKit/OpenSatKit #24: cFS password not accepted

gstuart-mei, 2020-02-13:
Hi,
I just downloaded and installed OpenSatKit. When I click on 'Start cFS' I get the terminal window. However, it won't accept the password. I have tried: 'osk', 'OSK', 'Osk', and just return.
Thanks

hstankey, 2020-03-10:
Having this issue as well. Have you managed to get a solution?

gstuart-mei, 2020-03-10:
Yeah, run it at su. 
However, the real problem is that the cfs executable doesn't exist; you need to go to that folder and build it.
Good luck!

dmccomas, 2020-03-11:
I apologize for the slow response. I retired from NASA and thought I had replaced my NASA email associated with this account. 

I think I found the problem. What's happening is the cFS is crashing.  Looks like I mistakenly CM'd the wrong es_startup.scr file.  This file identifies which apps to load during startup.   The Health & Safety (HS) app should not be loaded. It crashes on Ubuntu 18.04 and I haven't had time to debug it. 
es_startup.scr  is defined in /cfs/osk_defs/cpu1_es_startup.scr.  This file is renamed to es_startup.scr and copied into cfs/build/exe/cpu1/cf during the make/install process. 
 
To prevent HS from being loaded comment out the line below by putting an exclamation at the start of the line

CFE_APP, /cf/hs.so, HS_AppMain, HS, 120, 16384, 0x0, 0;

You can edit  cfs/build/exe/cpu1/cf /es_startup.scr  directly to quickly see if it's the problem. Just don't forget to change the cpu1_es_startup.scr 

Let me know if this fixes the issue.

Natpanzarini, 2020-06-08:
Hello!

I have tried:

> To prevent HS from being loaded comment out the line below by putting an exclamation at the start of the line CFE_APP, /cf/hs.so, HS_AppMain, HS, 120, 16384, 0x0, 0;

but that line already begins with an exclamation mark in my /cfs/osk_defs/cpu1_es_startup.scr file. I notice that my path is not exactly what is shown above, is there something else I need to do to get the password to be recognized?

gstuart-mei, 2020-06-08:
FYI, when I had this issue, the problem turned out to be simply that I hadn’t compiled CFS.