OpenSatKit/OpenSatKit #27: CFS Compilation Error

DKlaren, 2020-03-04:
I'm having trouble compiling the latest version of master on Ubuntu 18.04. The compilation stops with the following error:

/home/duser/dev/OpenSatKit/cfs/apps/sbn/fsw/src/sbn_app.c:31:10: fatal error: cfe_sb_priv.h: No such file or directory
 #include "cfe_sb_priv.h" /* For CFE_SB_SendMsgFull */
          ^~~~~~~~~~~~~~~
compilation terminated.

dmccomas, 2020-03-11:
Sorry for lack of response. I retired from NASA and thought I had changed this notification email from my NASA email to my personal email.

SBN = Software Bus Network and this app is used for bridging the software bus across processors. If you don't this app then I recommend removing it from the target build list used by cmake.  You do this by removing "sbn" from the following line in cfs/osk_defs/targets.cmake

SET(TGT1_APPLIST osk_app_fw bm cfs_lib cs ds expat_lib fm osk_demo_app hc hs hsim kit_ci kit_sch kit_to lc md mm sc hk tftp i42 f42 isim cf sbn)
 
The rest of this information is to troubleshoot why SBN isn't building. First we need to confirm SBN's CMakeLists.txt file (in SBN app base directory) is configured properly. It should contain the lines as in the attached CMakeLists.txt file.

# workaround until these definitions are exposed by cfe_sb
include_directories(${CFECORE_SOURCE_DIR}/src/sb)

If you confirm this and it still doesn't work let me know, assuming you want to debug it. As I mentioned SBN is not required for a single processor system.

DKlaren, 2020-06-09:
Thanks for the reply I reverted to the version before you included CFE 6.7.1 and it seems to be working out so far. Thanks.

Natpanzarini, 2020-06-11:
> Thanks for the reply I reverted to the version before you included CFE 6.7.1 and it seems to be working out so far. Thanks.

Hi, i'm having trouble building cfs so I tried reverting to the version before CFE 6.7.1 as well. I'm still having the same build issues with cfs even with that version.

Did you build cfs, 42, and cosmos all separately after getting the previous version?