OpenSatKit/OpenSatKit #28: Unknown target CPP mapped to interface LOCAL_CFS_INT

eupharis, 2020-03-26:
Hi! I had this running before, but I'm running a fresh install on Ubuntu 18.04 and running into issues.

Maybe this is related to that commit from 15 days ago?

Screenshot:

[image]

Text:

```
Error at /home/daniel/oresat/OpenSatKit-master/cosmos/config/tools/cmd_tlm_server/cmd_tlm_server.txt:65

TARGET CPP

Unknown target CPP mapped to interface LOCAL_CFS_INT

Parsed output in /home/daniel/oresat/OpenSatKit-master/cosmos/outputs/tmp/config/tools/cmd_tlm_server/cmd_tlm_server.txt

For more information see http://cosmosrb.com/docs/system/#command-and-telemetry-server-configuration.

NOTE!: The application will exit once you accept or dismiss this dialog!
```

eupharis, 2020-03-26:
Ok! I got it running and "start CFS" now works by:

* running `make && make install` in the cfs directory
* commenting out "/home/daniel/oresat/OpenSatKit-master/cosmos/config/tools/cmd_tlm_server/cmd_tlm_server.txt:65"

This was after a fresh install on Ubuntu 18.04 using the script in the Readme:

bash <(\wget -qO- https://raw.githubusercontent.com/OpenSatKit/OpenSatKit/vendor/install.sh)

dmccomas, 2020-03-27:
Unfortunately this is the result of me rushing in a different change
without checking thins out.

You can fix the problem by removing the line "TARGET CPP" in the following
file:

.../cosmos/config/tools/cmd_tlm_server/cmd_tlm_server.txt

Sorry for the inconvenience. I'll get a fix out ASAP.

- Dave

eupharis, 2020-03-28:
Good to know! Thanks for the update!