OpenSatKit/OpenSatKit #30: Run Script -> Ops Example -> No Prompt?

eupharis, 2020-04-03:
99% of the app working perfectly. Woohoo! Running on Ubuntu 18.04.

Found another hopefully little bug with running the Ops script:  

[image]

When I select it, it just freezes. Then when I force quit, this pops up: 

[image]

I think maybe just no terminal is popping up to take my password?

Maybe you can just add a spawn xfce4-terminal when this particular script runs like in some other spots?

```
spawn("xfce4-terminal --default-working-directory=""#{Osk::CFS_EXE_DIR}"" --execute sudo ./core-cpu1""")
```

dmccomas, 2020-05-12:
If the cFS is not running when you start the simsat ops example a message box should be displayed stating the cFS will be started and a new terminal window should appear. It's the same script that is used to start the cFS that is used by the main window. 

Are you running everything locally?

eupharis, 2020-05-12:
Yup, running everything locally!

Ok now everything is working great.

It looks like the cause was that sometimes I have some "extra" `./core-cpu1` processes that hang around after playing around with OpenSatKit for awhile, even after trying to hit `Stop CFS`.

[image]

If I make sure those are all dead and then do a fresh `ruby Launcher` for cosmos, then the ops script works great! Thanks!

closed 2020-05-12 by eupharis

dmccomas, 2020-05-28:
I changed the 'Stop cFS" so it usses an executive service command thinking
it's more graceful but I may go back to the old way that stops all
core-cpu1 processes.