OpenSatKit/OpenSatKit #31: Install error on Ubuntu 20.04

racerxdl, 2020-05-23:
For some reason there is a invalid replacement on line 16:

```
=============================================================================================================================================================================================================
INSTALLATION WILL TAKE TIME AND INCLUDES PROMPTS!
=============================================================================================================================================================================================================

./install.sh: line 16: OPEN_SAT_KIT_PATH: invalid direct expasion
```

That only happens on ubuntu 20.04, I installed a 18.04 on vm and it worked.

rkhanna19, 2020-06-01:
I have the same issue except my response looks like this:
```
===============================================================================
INSTALLATION WILL TAKE TIME AND INCLUDES PROMPTS!
===============================================================================

/dev/fd/63: line 16: OPEN_SAT_KIT_PATH: invalid indirect expansion
```

Remakh, 2020-06-03:
Hi I belive I solved it. There is a type in the installation file that is used for this install https://raw.githubusercontent.com/OpenSatKit/OpenSatKit/vendor/install.sh.

Line 16 of this file:

 if [ ! -n "${!OPEN_SAT_KIT_PATH}" ]; then

Should instead be: 

if [ ! -n "${OPEN_SAT_KIT_PATH}" ]; then

I copied the installation file into a text file and saved it into a bash script called install.sh. After that I ran it after changing the line and it is working fine.

dmccomas, 2020-06-12:
Thanks!  I made the update along with two chmod additions to give execution
privilege to the 42 executable and to cmake.sh that is used in the build
process for the create 'hello world' app process.

dmccomas, 2021-02-26:
There are additonal issues.  COSMOS is not currently compatible with Ubuntu 20.04. See COSMOS issues [1169](https://github.com/BallAerospace/COSMOS/issues/1169) and [1171](https://github.com/BallAerospace/COSMOS/issues/1171) for details. COSMOS 5.0 is in a beta test release and it's making an architectural departure from previous COSMOS versions.

It would be desirable to have OSK work with COSMOS 4.x on Ubuntu 20.04. Thanks to Jim Johansen for the following fix:
sudo add-apt-respository ppa:rock-core/qt4
sudo apt-get update
sudo apt-get install libqtcore4
bundle install

Updating OSK to work with COSMOS 5.x when its ready is a separate issue...

dmccomas, 2022-05-24:
Ubuntu 20.04 LTS is now supported starting with OSK v2.9!. COSMOS 4.* became incompatible with Ubuntu 20.04. See COSMOS issues [1169](https://github.com/BallAerospace/COSMOS/issues/1169) and [1171](https://github.com/BallAerospace/COSMOS/issues/1171) for details. OSK's installation script was updated in v2.9 to overcome these issues by installing a libqt4 Personal Package Archive (PPA) repo rock-core. Note this PPA has not been formally validated.

closed 2022-05-24 by dmccomas

mdicicco, 2023-01-13:
Has anyone tried this recently?  I was unable to get the install script to succeed.  The bundle install was trying to install cosmos 5.0.5, the latest version on rubygems.org.  I was able to add a '4.5.2' specifier to the Gemfile which does get cosmos 4 and attempts to install qtbindings, but that seems to have a build error on the latest 20.04.