OpenSatKit/OpenSatKit #35: Tlm Packets error

kihwankim, 2020-07-09:
Hi! I followed this video (https://www.youtube.com/watch?v=LspTUdIKAZY&t=36s)

when I click here
[image]

and

[image]

and this is my cMake and cpu1_cfe_es_startup.scr

[image]

I can not find why it happened

dmccomas, 2020-07-09:
Thanks for the details, they're very helpful.

It looks like the file ~/cosmos/config/tools/cmd_tlm_server/
cmd_tlm_server.txt was not automatically updated.  The tool should have
inserted a line
"TARGET EXAMPLE" between the   <cfs_kit> and ,</cfs_kit> tags.  Hopefully
the following steps will fix the problem:

1. Add a "TARGET EXAMPLE" line to the list of targets under the "INTERFACE
LOCAL_CFS_INT udp_cs_interface.rb 127.0.0.1 1234 1235 nil nil 128 nil nil".
Between the <cfs_kit> tags is fine.
2. Restart COSMOS.

Is anything between the  <cfs_kit>  tags? I'm wondering what happened.

Thanks - Dave

kihwankim, 2020-07-13:
thank you for your answer.
[image]

but there's another problem.
[image]
[image]

I opened like this but It said the pakcet has never come. so I wonder what's wrong is there and if there is any other data needed to analyze the problem, I will upload the photo or log again.

thank you

kihwankim, 2020-07-13:
and Stop cFS/Server is not working 

[image]

in this page

thank you!!

ArrobaRicardoGE, 2021-02-20:
@kihwankim Were you able to solve the problem of the missing packet? I am having a similar poblem.

dmccomas, 2021-02-21:
Sorry this dropped off the radar.  Its confusing because EXAMPLE is showing up as an available target but the error states its not a target. I've tried a lot of permutations of things that may hav egone wrong during the automated process but I can't reporduce the issue. Can you please attach the exception log, hopefully some clues.

dmccomas, 2021-02-21:
I get the same error if the COSMOS Command and Telemetry Server is not restarted after the EXAMPLE code is generated.  It could be a problem with the script that runs when button 4 (Stop cFS/Server) is pressed.  

You can manually close the GUI titled "cFS Command and Telemetry Server" and on the COSMOS launcher start the server back up by selecting the "Command and Telemetry Server" button.  See pic although the button order has changed in later releases. Select 'OK' on the pop up that asks about 'cmd_tlm_server.txt'.  

If you have shut everthing down since first generating the code the probelm should not have persist when you restart OSK.  If it does then we have another problem.  

[image]

closed 2023-05-14 by kihwankim