OpenSatKit/OpenSatKit #40: Question: What Version of 42?

PeteBlanchard, 2020-08-18:
Hello, I was wondering what version of 42 you are using?  

I was looking at what is checked in and it appears you have a static copy checked out (sometime) from "https://mesa-svn.ndc.nasa.gov/42".  I know Mr. Stoneking has a [42 github](https://github.com/ericstoneking/42), but can't compare the versions very well...git <-> svn...

I was trying to change the configuration of the satellite and my example is from a (very) different version of 42.

Thanks in advance.

Regards,

  Pete

PeteBlanchard, 2020-08-18:
okay, found out by digging a bit (and installing subversion):

```bash
URL: https://mesa-svn.ndc.nasa.gov/42
Relative URL: ^/
Repository Root: https://mesa-svn.ndc.nasa.gov/42
Repository UUID: ad5bd6d9-799f-e640-a0ec-14e48dfc8bbe
Revision: 987
Node Kind: directory
Schedule: normal
Last Changed Author: kepeck1
Last Changed Rev: 987
Last Changed Date: 2017-11-15 13:35:24 -0800 (Wed, 15 Nov 2017) 
```

Change in question...Is there any plan on updating the version?  Do you have a roadmap?

timothy, 2020-08-18:
Updating 42 to a newer version would be great. There is a lot of extra functionality in the latest version of 42 that OSK is missing out on. I would be interested in knowing if there are any plans to update as well.

dmccomas, 2020-08-20:
It's been on my list to upgrade 42. The last time I did it was a significant imapct to the 42 interface and controller apps.  The reason it was an impact is because I was trying to show how the same controller in 42 could be migrated to the FSW.  My guess is the 42 interface app will be impacted but I can always leave the controller alone. 

I'll bump the priority. I'll also talk to Eric about tagged versions so I can track things better.

PeteBlanchard, 2020-08-20:
Thanks for the response, David.

I did a quick sub-out with the new 42 and it looks like there have been some changes...looks like the 42fsw.c has been re-vamped and some methods changed.  I know Eric changed how the IO sockets worked fairly recently as well (you may not need your custom comm code anymore).  But I really didn't have much time to dig into you code (I saw that the three axis was hard-coding params of SC_Aura.txt)...

I had hoped to change the base config of the sim to use a small satellite in a software testing environment that simulates a CubeSat satellite and its associated subsystems.

dmccomas, 2020-10-20:
Hi Pete,

Well the priority bump was relative to a full plate but I was finally able to get it done. I released OSK v2.4today and it includes 42 release '2042'.  There were substantially changes especially in the extrenal interface design. Here's my [release blog](https://opensatkit.github.io/journal/Version-2_4-Released.html). Not quite a version description document but hopefully helpful.  There are notes about how I changed the 42 controller implemented in the cFS F42 app.  

Let me know if you have any questions.  I would like to make F42 a little more realistic from a more complex mission and ops perspective.  Now that 42 has a better defined external interface it may be a good time to not simple rehost a 42 controller in F42 but to make some addiitons.  One caveat is that I'm a software engineer not an aersopace engineer. 

- Dave

PeteBlanchard, 2020-10-20:
Awesome, thanks Dave.  I know there were substantial changes with 42 (additional sensor types as well as the re-structuring of I/O in IPC).

closed 2020-10-20 by PeteBlanchard