OpenSatKit/OpenSatKit #44: OSK cFS System Time Fails to Start

japalmerct1971, 2020-11-28:
Hello,
  I'm learning the cFS with OpenSatKit running on a Raspberry Pi 4 with Ubuntu 18.4 LTS. Following the Hello World app tutorial, when I click "Start cFS" in the Explore cFS/SimSat tab of the CFS_KIT screen, the terminal window does not come up and the System Time field does not turn white.  System time also fails to start if I click the Enable_Tlm config command.  Can you suggest how to remedy this error, please?

dmccomas, 2020-11-30:
I'm a little baffled that you don't see any terminal window.  Here are some
thoughts:

1. After you try "Start cFS" and then you shut down all open OSK/COSMOS
windows do you see any hidden error dialogue boxes?

2. Can you open a new Ubuntu terminal window and enter "xfce4-terminal" at
the command prompt. You should see a new window be created. This confirms
xfce4-terminal is installed which is the program that OSK uses to run the
cFS.

3. This doesn't explain no terminal, but I'm wondering whether the cFS
built properly during installation and the cfe executable doesn't exist.
Would you please check to see whether the following file exists:
~/OpenSatKit-master/cfs/build/exe/cpu1/core-cpu1  I tested this situation
on my setup and the "Start cFS" terminal window still appears but
immediately disappears when you enter a password which makes sense since it
can't locate the cfe-cpu1 so this doesn't explain no window but it is a
general sanity check.

If we can't get any further via email, I'd be happy to have a video call
where you could share your screen and I can observe what's going on.

Thanks - Dave

japalmerct1971, 2020-12-02:
Dave:

Many thanks for this prompt response and these thoughtful suggestions!
After investigating, I found that there are no hidden error dialogue boxes
and the xfce4-terminal command does indeed open a new terminal window.

What's very interesting is that when I look for core-cpu1.exe, I can't find
it.  In fact, ~/OpenSatKit-master/cfs/build is all that I have in common
with the directory path listed above: the exe directory does not exist.
I'm not certain what I would do differently to remedy this in a
re-installation. I appreciate your help.

Very respectfully,

- JP

dmccomas, 2020-12-02:
That indicates the cFS image wasn't built.  If you open a terminal window
and go to  ~/OpenSatKit-master/cfs/  then enter

make prep
make install

If you get any error messages then something didn't install correctly
during the installation process and we can figure where to go next based on
the errors.

- Dave

japalmerct1971, 2020-12-05:
Dave:

  Many thanks for this reply.  Here's what happens when I enter make prep
from ~/OpenSatKit-master/cfs/:

-- Configuring for system arch: cpu1
-- The C compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52
(message):
  The C compiler

    "/usr/bin/gcc"

  is not able to compile a simple test program.

I read online some comments indicating that this error is caused by using
an incompatible version of the GNU Compiler Collection (gcc) with Ubuntu
18.04.  So, I installed gcc-8 and configured the system to use it.  No
dice.  As you can see, I still have the same problem.  I appreciate your
guidance and happy holidays!

V/R,

J. Palmer