OpenSatKit/OpenSatKit #46: Question: Does the OpenSatKit has a communication port?

pitachan23, 2021-01-20:
I'm currently trying to develop a web application that can communicate with the OpenSatKit. Does the OpenSatKit have a communication port (e.g TCP/IP, etc.) where I can send command to/receive command from the OpenSatKit application? Any help appreciated!

dmccomas, 2021-01-21:
Two apps use UDP to communicate with the COSMOS ground system or with any
external system.
   Command Ingest, KIT_CI,
https://github.com/OpenSatKit/OpenSatKit/tree/master/cfs/apps/kit_ci/fsw/src
receives commands from a UDP socket and sends the commands on the Software
Bus to other apps
   Telemetry Output, KIT_TO,
https://github.com/OpenSatKit/OpenSatKit/tree/master/cfs/apps/kit_to reads
packets from the Software Bus from apps and sends them to a UDP socket to
an external system

I can help you dig through the code if needed.

I'm curious about your web application. I'm investigating
containerizing the cFS and using some web interfaces to access it.

- Dave

pitachan23, 2021-01-26:
We have developed a blockchain based web application for satellite configuration. The main purpose is to try and send satellite parameters to the ground command. Our parameters currently consist of sensor information and classical orbital elements (such as inclination, eccentricity, RAAN, mean anomaly, etc.), but the list of parameters can be modified to match configuration parameters needed by OpenSatKit. After looking through the documentation, I think we should be interfacing with COSMOS rather than cFS. If you can give us any pointers as to what kind of configuration COSMOS has, we'd really appreciate. Thank you so much!

- Pita

closed 2021-03-30 by pitachan23

dmccomas, 2021-03-31:
I just found this reply in my draft folder!! Thought I'd send it even
though its 2 months old...

Sounds like an interesting project.  I'll describe a few things and
hopefully one will help

1. The FSW command ingest (kit_ci) and telemetry output (kit_to) apps
communicate with COSMOS via UDP so you could talk directly to these apps
except you'd have to encode/decode the binary packets and that's what
COSMOS is good at.

2. My background is the cFS so I'm not a COSMOS expert. I think the COSMOS
JSON API may be helpful. https://cosmosrb.com/docs/v4/json-api. I've never
used it but it looks like you get a remote symbolic cmd/tlm interface

3. NASA maintains what I'll call a mini groundn system
https://github.com/nasa/cFS-GroundSystem/blob/10efa7050f82a4acc3013e9fddf31fd0ea5efe28/Guide-GroundSystem.md

I haven't used it but the ZeroMQ has the following description that looks
promising:

"The GroundSystem directory contains the new Ground System project for the
cFS, that incorporates the main window to launch commands and telemetry
systems, and other utilities like FDL/FUL and FT managers to send and
receive files. The main window runs alongside the Routing Service (
RoutingService.py). The Routing Service handles all incoming data and
publishes (PUB/SUB) the data to specific ZeroMQ channels so that the
different ground system utilities can receive (subscribe) only the desired
data."

Hope this helps.  - Dave