OpenSatKit/OpenSatKit #49: Error when starting up cFS in the terminal. Status = -17 (OS_ERR_NAME_NOT_FOUND)

ChrisPasquinelli, 2021-01-28:
When starting cFS in the terminal I get an error saying ****EVS Port1 42/1/CF 73: SemGetId Err:Chan 0 downlink PDUs cannot be throttled.0xFFFFFFEF**** After **1980-012-14:03:21.97063 ES Startup: CFE_ES_Main entering OPERATIONAL state**. 

Ran it through the debugger to  track where the error was coming from and seems like the error gets printed while running cf_app.c file in the CF_GetHandshakeSemIds() function.

dmccomas, 2021-01-29:
You nailed it!

File transfer is something we should discuss. It was brought up during the
Dellingr but that was a little early.

dmccomas, 2021-01-29:
That event message is from the CFDP app and it's expected in the OSK
configuration but not documented.  In a flight system the CF app shares a
semaphore with the telemetry output app and they use it to prevent queue
overflows into TO app.  Currently the TFTP app is the default file
transfer app in OSK.  In OSK v2.5 CF was added and it is loaded and
available but not used by OSK when a file transfer request is made from
COSMOS.  I'll look into adding the CF semaphore support in KIT_TO.