OpenSatKit/OpenSatKit #58: OpenSatKit Start cFS not doing anything

SadiqSarwar, 2021-03-02:
When pressing OpenSatKit through the launcher I only see the CFS_KIT_SCREEN window. The Telemetry Viewer and CFS Command & Telemetry Server are not visible, but show as open in my taskbar. Pressing Start cFS opens the terminal for me to enter my password, but upon entering my password, the terminal closes immediately and nothing happens. The System Time does not increment either and stays as 0.

dmccomas, 2021-03-03:
I'm wondering whether the COSMOS windows are off the display.  Are you running in a VM?  If so you may need to adjust your display resolution. 

For the terminal closing immediately it may be that the cFS wasn't built during installation. Would you please check to see whether the following file exists: ~/OpenSatKit-master/cfs/build/exe/cpu1/core-cpu1

SadiqSarwar, 2021-03-09:
Here are the files that I see when I open the ~/OpenSatKit-master/cfs/build/exe/cpu1/ directory
[image]

Regarding the cFS build, I think you might be right. Here is what I get after running make prep and then make install:
[image]

Do you know what may be the issue?

dmccomas, 2021-03-16:
I'm not sure why the Checksum (CS) app table build is failing. Let's try
removing CS from the build to see whether everything else builds. To remove
CS from the build open the following file in a text editor:

~/OpenSatKit-master/cfs/osk_defs/targets.cmake

Locate the following line. It should be around line 53. Remove the 'cs'
entry.

SET(TGT1_APPLIST cfs_lib expat_lib osk_c_fw osk_cpp_fw osk_42_lib mqtt_lib
kit_ci kit_sch kit_to cs ds fm hs lc md mm sc hk tftp cf i42 f42 isim bm hc
hsim filemgr osk_c_demo osk_cpp_demo mqtt)

In a terminal window change directory to  ~/OpenSatKit-master/cfs  and issue

make prep
make install

Hopefully this will work. I just added CS back into the default build for
version 2.8 so maybe something still isn't right.

- Dave