OpenSatKit/OpenSatKit #60: Question: Sending a command to cFS

pitachan23, 2021-03-08:
I tried sending a string command to UDP server on localhost port 1234 with the script below while cFS is running:

require 'socket'
s = UDPSocket.new
r = s.send("hello", 0, 'localhost', 1234)

Here is the message I got from core Flight System terminal:
EVS Port1 42/1/CFE_SB 21: Send Err: Invalid MsgId(0x6865) in msg.App KIT_CI

-------------------------------------------------------------------------------------------
I know that I'm sending a dummy command, but when I send string "NOOP" I also get a similar output message from cFS. I have two questions:
1) When I send command this way, is the command being sent directly to KIT_CI or cFS?
2) Does the command have to be different type other than string (i.e. byte)? What commands am I allow to send directly to KIT_CI or cFS and bypass COSMOS?

Ultimately I would like to be able to send a command to retrieve and update configuration table similar to what TableService demo shows, but right now I'm just trying to send simple command like NOOP first.

Thank you,
Pita

dmccomas, 2021-03-09:
Your question made me realize I don't have the COSMOS-to-cFS command flow documented in a single concise place.  Good topic for a future video!  Hopefully I can string the pieces together in a meaningful way here.

1.   Binary packets are transferred between COSMOS and KIT_CI (Command Ingest) which is an app running in the cFS.  The term cFS is used to refer to all of the FSW running. The binary packet format is defined on slide 5 in the following Software Bus slides.   
https://github.com/OpenSatKit/OpenSatKit/blob/master/cosmos/cfs_kit/tutorials/cfe/training/OSK-T02-1-cFE-SB.pdf
The same CCSDS packet format used between ground and flight is used onboard.

2. The first 2 bytes of the message is the ID so in your example KIT_CI tried to interpret 'he' as the message ID which corresponds to the 0x6865 in the event message. 0x68 = ASCII 'h' and 0x65 = ASCII 'e'

3.  COSMOS takes care of creating the binary command packet when you send a command from either a screen or from a COSMOS script that is run from the script runner or test runner tool.  COSMOS creates the binary packet using the command definitions that are defined in "targets". Every FSW app has  COSMOS target that defines the app's command and telemetry packets. For example the following directory contains KIT_CI's command and telemetry definitions: https://github.com/OpenSatKit/OpenSatKit/tree/master/cosmos/config/targets/KIT_CI/cmd_tlm

4. Here's KIT_CI's NOOP command in the COSMOS Command Sender tool. The CCSDS fields correspond to the diagram in the SB slides I mentioned above. 

[image]

5. When you send a command the Command Sender captures the COSMOS script command which looks like this for the KIT_CI NOOP and this is the syntax a script can use. 
cmd("KIT_CI NOOP with CCSDS_STREAMID 7936, CCSDS_SEQUENCE 49152, CCSDS_LENGTH 1, CCSDS_CHECKSUM 0, CCSDS_FUNCCODE 0")

6. The Runtime Environment App video may also be helpful. It describes parts of this as well
https://www.youtube.com/watch?v=p3RrovUyLOs

Let me know if have more questions.

pitachan23, 2021-03-09:
Thank you for the explanation. That helps clear up some of my confusion.

In number 5, when you mentioned "this is the syntax a script can can use." Which script are you referring to here? 

Is there a way I can use the command below without having to do it through the GUI, but running it from a script instead?

     cmd("KIT_CI NOOP with CCSDS_STREAMID 7936, CCSDS_SEQUENCE 49152, CCSDS_LENGTH 1, CCSDS_CHECKSUM 0, CCSDS_FUNCCODE 0")

Thank you

pitachan23, 2021-03-09:
What I'm hoping to achieve is sending a command from my web application to cFS. I'm just not sure which would be the best way to do it. At first I thought I would be able to send a command directly to cFS through UDP socket but it seems that I still have to translate the command to binary CCSDS format, which is handled by COSMOS. I'm not looking to reproduce something that is already done by COSMOS, but I'm wondering if it will be possible for me to write a ruby script that utilizes some COSMOS libraries to create a binary command packet and send to cFS without having to use any GUI tool (e.g. script runner or test runner tool)?

dmccomas, 2021-03-10:
COSMOS scripts are run in either the Script Runner or Test Runner.  For now stick with Script Runner because Test Runner involves a framework that will be helpful when you're ready to write tests but it adds a layer you don't need yet.  The COSMOS scriptng guide is at https://cosmosrb.com/docs/v4/scripting.  COSMOS has added a lot of methods like cmd() on top of Ruby. 

I started Sript Runner from the Launcher and pasted the KIT_CI NOOP command and hit start and it executed it. Ignore the yellow warning about bit definitions in the Script Runer output box, I'm working that issue.

[image]

Here's an example OSK script that demos some File Manager app features that you can run in Script Runner

https://github.com/OpenSatKit/OpenSatKit/blob/master/cosmos/config/targets/FM/procedures/demo_fm_features.rb

As a heads up, OSK adds a thin layer on top of COSMOS's scripting API. For example the FM script contains lines like the one below. These run methods that do common processing like verify a valid command counter in telemetry after a command is sent. You don't have to use OSK's layer and the OSK/cosmos/lib code serves as the documentation at the moment.

Osk::flight.send_cmd("FM","SEND_DIR_PKT with DIRECTORY #{Osk::FLT_SRV_DIR}, DIR_LIST_OFFSET 0")

pitachan23, 2021-03-12:
Hi David,

Thank you so much for your help. I have what I need now.

Thank you,
Pita

closed 2021-03-12 by pitachan23