OpenSatKit/OpenSatKit #62: How to start cFS without the GUI?

pitachan23, 2021-03-29:
Hi, Is there a way to start cFS from a command line without starting from the OpenSatKit GUI? 

Additionally, when I send a command (e.g. NOOP to KIT_CI) Open Sat Kit GUI flight event messages show "Kit Command Ingest (KIT_CI) version 1.1.0 received a no operation command." Are flight event messages saved to log file somewhere?

Thank you

dmccomas, 2021-03-30:
Yes you can start the cFS from a command line.  If you open a terminal
window and change directory to

~/OpenSatKit-master/cfs/build/exe/cpu1

then run

./core-cpu1

it will start the cFS. It looks for the CFE_ES startup script to identify
which apps to load

~/OpenSatKit-master/cfs/build/exe/cpu1/cf/cfe_es_startup.scr

The cFE event service stores events in a ram buffer that can be written to
a file by commanding CFE_EVS to write the buffer to a file. If the COSMOS
GUI is too heavy weight the cFS Bundle comes with a python GUI.  You will
need some front end to command the cFS services/apps.

- Dave

ryuxin, 2022-02-10:
I have one more question.
How can I start both cFS and 42 without GUI? In my understanding, cFS and 42 interact with each other via socket, so I need both to run together.
Thanks.