OpenSatKit/OpenSatKit #65: PerfUtils-java not working

mogar, 2021-04-13:
I'm trying to run perfutils-java, but having dependency issues.

I've downloaded jfreechart-1.0.18, compiled it with ant, and am trying to run the CPM.jar file as follows:

```
java -cp ~/jfreechart-1.0.18/lib/jfreechart-1.0.18.jar:~/jfreechart-1.0.18/lib/jcommon-1.0.22.jar -jar CPM.jar
```

I get the following errors:

```
Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: org/jfree/data/general/PieDataset
	at CFSPerformanceMonitor.CPMMain.initialize(CPMMain.java:588)
	at CFSPerformanceMonitor.CPMMain.<init>(CPMMain.java:249)
	at CFSPerformanceMonitor.CPMMain.<init>(CPMMain.java:227)
	at CFSPerformanceMonitor.CPMMain$1.run(CPMMain.java:313)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.ClassNotFoundException: org.jfree.data.general.PieDataset
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:471)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	... 17 more

```

I also notice that the documentation says that this tool is intended to work with cFE 6.3. Has anyone gotten it working with cFE 6.8?

dmccomas, 2021-04-14:
You should be able to just run the jar without any additional libraries.
"java -jar CPM.jar"

A while back (order of years) there was a jfreechart library version issue
with a deprecated API. Your error rings that bell The creator of the tool
made an update that's included in OSK. Are you running with a fairly recent
version of OSK?

- Dave

mogar, 2021-04-14:
I just cloned from Master yesterday, so I should be up to date unless there's a different branch I should be using.

I tried running without installing any other libs, and got a slightly different error. Then I read through the docs* which said that JFreeChart was a pre-req.

I have seen PerfUtils-win working, but I'm developing on a Linux machine so that would be pretty inconvenient to use.

* These docs: https://github.com/OpenSatKit/OpenSatKit/blob/master/cfs/tools/perfutils-java/src/docs/CPM_Users_Guide.pdf

dmccomas, 2021-04-14:
Glad you sent the link. That is the version that comes with the cFS which I
didn’t realize has the old problem. I maintain a CPM under the cosmos dir
tree.  Try OSK/cosmos/cfs_kit/tools/perfmon I think that’s the path I have
to run...

mogar, 2021-04-15:
That works for me, thanks!

mogar, 2021-05-04:
Is it possible to get access to the source code for CPM.jar? We want to adapt it slightly to make it scriptable.

mogar, 2021-05-05:
Nevermind. I just realized they were right there in that directory :P

closed 2021-05-05 by mogar