OpenSatKit/OpenSatKit #72: Error starting osk_42_lib.so and KIT_TO from cFS FSW Edu tab

igor10091972, 2021-07-13:
Hi all!

I'm facing the following:
OSK starts normally when I press the Start cFS button on the Missions FSW tab: the System Time value increases and in the Cmd/Tlm server application, the LOCAL_CFS_INT line increases the number of bytes in the Byte Rx column. But when I try to do the same on the cFS FSW Edu tab, the System Time value does not change and the number of bytes in the Byte Rx column does not change. In this case, the terminal contains the following errors:

> ...
> 1980-012-14: 03: 20.25673 ES Startup: Loading shared library: /cf/osk_42_lib.so
> Error loading shared library: ./cf/osk_42_lib.so: cannot open shared object file: No such file or directory
> 1980-012-14: 03: 20.25688 ES Startup: Could not load cFE Shared Library: -1
> 1980-012-14: 03: 20.25734 ES Startup: Loading file: /cf/kit_to.so, APP: KIT_TO
> Error loading shared library: ./cf/kit_to.so: undefined symbol: round
> 1980-012-14: 03: 20.25754 ES Startup: Could not load cFE application file: /cf/kit_to.so. EC = 0xFFFFFFFF
> ....

Attempting to click Send Config Cmd Enable Telemetry nothing to do.

I would be grateful for any help or suggestions on how to try and fix these errors.

Igor

OS description: Ubuntu 18.04.5 LTS

Just in case, I will give the terminal full content below:
> [sudo] password for kepler: 
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 2
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cfsat
CFE_PSP: Starting the cFE with a POWER ON reset.
Policy 1: available, min-max: 1-99
Policy 2: available, min-max: 1-99
Selected policy 2 for RT tasks, root task = 99
Maximum user msg queue depth = 10
OSAL: Registering VolumeTable map for ./cf on /cf
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2031-194-12:41:33.12249 POWER ON RESET due to Power Cycle (Power Cycle).
2031-194-12:41:33.12255 ES Startup: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
2031-194-12:41:33.12260 ES Startup: CFE_ES_Main entering CORE_STARTUP state
2031-194-12:41:33.12260 ES Startup: Starting Object Creation calls.
2031-194-12:41:33.12261 ES Startup: Calling CFE_ES_CDSEarlyInit
2031-194-12:41:33.12304 ES Startup: Calling CFE_EVS_EarlyInit
2031-194-12:41:33.12311 Event Log cleared following power-on reset
2031-194-12:41:33.12312 ES Startup: Calling CFE_SB_EarlyInit
2031-194-12:41:33.12346 SB internal message format: CCSDS Space Packet Protocol version 1
2031-194-12:41:33.12354 ES Startup: Calling CFE_TIME_EarlyInit
1980-012-14:03:20.00000 ES Startup: Calling CFE_TBL_EarlyInit
1980-012-14:03:20.00019 ES Startup: Calling CFE_FS_EarlyInit
1980-012-14:03:20.00043 ES Startup: Core App: CFE_EVS created. App ID: 0
EVS Port1 42/2/CFE_EVS 1: cFE EVS Initialized. cFE Version 6.7.1.0
EVS Port1 42/2/CFE_EVS 14: No subscribers for MsgId 0x808,sender CFE_EVS
1980-012-14:03:20.05103 ES Startup: Core App: CFE_SB created. App ID: 1
1980-012-14:03:20.05150 SB:Registered 4 events for filtering
EVS Port1 42/2/CFE_SB 1: cFE SB Initialized
EVS Port1 42/2/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_SB
1980-012-14:03:20.10211 ES Startup: Core App: CFE_ES created. App ID: 2
EVS Port1 42/2/CFE_ES 1: cFE ES Initialized
EVS Port1 42/2/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/2/CFE_ES 2: Versions:cFE 6.7.1.0, OSAL 5.0.1.0, PSP 1.4.0.0, chksm 32145
EVS Port1 42/2/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/2/CFE_ES 91: Mission osk
EVS Port1 42/2/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/2/CFE_ES 92: Build 202107131541 kepler@ubuntu
1980-012-14:03:20.15317 ES Startup: Core App: CFE_TIME created. App ID: 3
EVS Port1 42/2/CFE_TIME 1: cFE TIME Initialized
1980-012-14:03:20.20386 ES Startup: Core App: CFE_TBL created. App ID: 4
EVS Port1 42/2/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.7.1.0
[... 114 lines cut ...]
1980-012-14:03:20.25954 ES Startup: Loading file: /cf/md.so, APP: MD
1980-012-14:03:20.25973 ES Startup: MD loaded and created
EVS Port1 42/2/MD 52: MD Dwell Tbl verify results: good = 0, bad = 0, unused = 25
1980-012-14:03:20.26098 ES Startup: Loading file: /cf/mm.so, APP: MM
EVS Port1 42/2/MD 52: MD Dwell Tbl verify results: good = 0, bad = 0, unused = 25
1980-012-14:03:20.26124 ES Startup: MM loaded and created
EVS Port1 42/2/MM 1: MM Initialized. Version 2.4.1.0
EVS Port1 42/2/MD 52: MD Dwell Tbl verify results: good = 0, bad = 0, unused = 25
EVS Port1 42/2/MD 52: MD Dwell Tbl verify results: good = 0, bad = 0, unused = 25
1980-012-14:03:20.26288 ES Startup: Loading file: /cf/tftp.so, APP: TFTP
EVS Port1 42/2/MD 7: Dwell Tables Recovered: 0, Dwell Tables Initialized: 4
EVS Port1 42/2/MD 1: MD Initialized.  Version 2.3.3.0
1980-012-14:03:20.26306 ES Startup: TFTP loaded and created
EVS Port1 42/2/TFTP 111: Socket bind failed. errno = 98
EVS Port1 42/2/TFTP 100: TFTP App Initialized. Version 1.1.0
1980-012-14:03:20.26348 ES Startup: Loading file: /cf/kit_sch.so, APP: KIT_SCH
1980-012-14:03:20.26369 ES Startup: KIT_SCH loaded and created
EVS Port1 42/2/KIT_SCH 25: Successfully replaced table 0 using file /cf/osk_sch_msg_tbl.json
EVS Port1 42/2/KIT_SCH 25: Successfully replaced table 1 using file /cf/osk_sch_sch_tbl.json
EVS Port1 42/2/KIT_SCH 101: KIT_SCH Initialized. Version 2.0.0
1980-012-14:03:20.36394 ES Startup: CFE_ES_Main entering APPS_INIT state
1980-012-14:03:20.36395 ES Startup: CFE_ES_Main entering OPERATIONAL state
EVS Port1 42/2/CFE_TIME 21: Stop FLYWHEEL
EVS Port1 42/2/KIT_SCH 406: Multiple slots processed: slot = 0, count = 2
EVS Port1 42/2/KIT_SCH 404: Major Frame Sync too noisy (Slot 1). Disabling synchronization.

igor10091972, 2021-07-13:
Oh, I seem to have solved my trouble!

I just added the osk_42_lib to the list of applications for the cFS FSW Edu tab in source file ~/OpenSatKit-master/cfs/osk_defs/targets.cmake.
Thus, in order for OSK to start normally from the cFS FSW Edu tab, the edited list of applications should look like this:

SET (TGT2_APPLIST cfs_lib osk_42_lib osk_c_fw kit_ci kit_sch kit_to tftp fm md mm).

And if the connection to the Cmd / Tlm server does not automatically occur, the System Time does not start and the value of the LOCAL_CFS_INT packet counter in the Bytes Rx column does not start to increase, then you must click the Send Config Command <Enable Telemetry> button.

Sincerely, Igor.

dmccomas, 2021-07-15:
Glad you were able to quickly work around the issue. My intention is to not have the 42 library loaded in the cFS education target so cfs/osk_defs/cfsat_cfe_es_startup.scr should not include the following line that loads the library

         CFE_LIB, /cf/osk_42_lib.so,   OSK_42_LIB_LibInit,  OSK_42_LIB,     0,   16384, 0x0, 0;

Here's a picture of the cfsat target.

[image]

I'm trying to get v3.1 out ASAP and it will have improved 'quick start' documentation for each target. And hopefully no bugs :-)

dmccomas, 2021-07-16:
When I removed osk_42_lib.so from the startup script I found another latent bug. KIT_TO relies on the round() function in the standard math libraries so its cmakelist.txt file should have the math library specified and it doesn't.  osk_42_Lib does include the math libraries so kit_to coudl dynamically link to the math libaries but when I removed osk_42_lib kit)to failed to load.  

In kit_to's cmakelist.txt I added the following line be below the line "add_cfe_app(kit_to ${APP_SRC_FILES})"
target_link_libraries (kit_to m)

igor10091972, 2021-07-18:
Hi, David!
I express my deep gratitude to you for your important work and your
reaction to my messages!
I have another question for you about the following messages that appear
several times in the terminals of both Mission FSW and cFS FSW Edu tabs and
what they mean:
...
EVS Port1 42/2 / CFE_SB 14: No subscribers for MsgId 0x808, sender CFE_SB
...
?

igor10091972, 2021-07-19:
And more ... When transferring and running VM OSK to another, more powerful
computer
even with a non-recompiled project, there are many errors like:

... Pipe Overflow, MsgId 0x808, pipe KIT_TO_PKT_PIPE ...

from different applications.
With what it can be connected?
I attach the file with the contents of the terminal ...

igor10091972, 2021-07-19:
And more ... When transferring and running VM OSK to another, more powerful
computer
even with a non-recompiled project, there are many errors like:

... Pipe Overflow, MsgId 0x808, pipe KIT_TO_PKT_PIPE ...

from different applications.
With what it can be connected?
I attach the file with the contents of the terminal ...

[sudo] password for kepler: 
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: simsat
CFE_PSP: Starting the cFE with a POWER ON reset.
Policy 1: available, min-max: 1-99
Policy 2: available, min-max: 1-99
Selected policy 2 for RT tasks, root task = 99
Maximum user msg queue depth = 10
OSAL: Registering VolumeTable map for ./cf on /cf
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2031-200-13:37:09.62158 POWER ON RESET due to Power Cycle (Power Cycle).
2031-200-13:37:09.62172 ES Startup: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
2031-200-13:37:09.62177 ES Startup: CFE_ES_Main entering CORE_STARTUP state
2031-200-13:37:09.62178 ES Startup: Starting Object Creation calls.
2031-200-13:37:09.62179 ES Startup: Calling CFE_ES_CDSEarlyInit
2031-200-13:37:09.62202 ES Startup: Calling CFE_EVS_EarlyInit
2031-200-13:37:09.62211 Event Log cleared following power-on reset
2031-200-13:37:09.62212 ES Startup: Calling CFE_SB_EarlyInit
2031-200-13:37:09.62232 SB internal message format: CCSDS Space Packet Protocol version 1
2031-200-13:37:09.62240 ES Startup: Calling CFE_TIME_EarlyInit
1980-012-14:03:20.00000 ES Startup: Calling CFE_TBL_EarlyInit
1980-012-14:03:20.00036 ES Startup: Calling CFE_FS_EarlyInit
1980-012-14:03:20.00065 ES Startup: Core App: CFE_EVS created. App ID: 0
EVS Port1 42/1/CFE_EVS 1: cFE EVS Initialized. cFE Version 6.7.1.0
EVS Port1 42/1/CFE_EVS 14: No subscribers for MsgId 0x808,sender CFE_EVS
1980-012-14:03:20.05112 ES Startup: Core App: CFE_SB created. App ID: 1
1980-012-14:03:20.05122 SB:Registered 4 events for filtering
EVS Port1 42/1/CFE_SB 1: cFE SB Initialized
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_SB
1980-012-14:03:20.10158 ES Startup: Core App: CFE_ES created. App ID: 2
EVS Port1 42/1/CFE_ES 1: cFE ES Initialized
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/1/CFE_ES 2: Versions:cFE 6.7.1.0, OSAL 5.0.1.0, PSP 1.4.0.0, chksm 32145
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/1/CFE_ES 91: Mission osk
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/1/CFE_ES 92: Build 202107180031 ***@***.***
1980-012-14:03:20.15237 ES Startup: Core App: CFE_TIME created. App ID: 3
EVS Port1 42/1/CFE_TIME 1: cFE TIME Initialized
1980-012-14:03:20.20279 ES Startup: Core App: CFE_TBL created. App ID: 4
EVS Port1 42/1/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.7.1.0
1980-012-14:03:20.25296 ES Startup: Finished ES CreateObject table entries.
1980-012-14:03:20.25301 ES Startup: CFE_ES_Main entering CORE_READY state
1980-012-14:03:20.25304 ES Startup: Opened ES App Startup file: /cf/cfe_es_startup.scr
1980-012-14:03:20.25358 ES Startup: Loading shared library: /cf/cfs_lib.so
CFS Lib Initialized.  Version 2.2.0.01980-012-14:03:20.25429 ES Startup: Loading shared library: /cf/osk_c_fw.so
OSK C Application Framework Library Initialized. Version 2.0.0
1980-012-14:03:20.25495 ES Startup: Loading shared library: /cf/osk_42_lib.so
OSK 42 Library Initialized. Version 1.0.0
1980-012-14:03:20.25600 ES Startup: Loading file: /cf/kit_to.so, APP: KIT_TO
1980-012-14:03:20.25635 ES Startup: KIT_TO loaded and created
1980-012-14:03:20.25686 ES Startup: Loading file: /cf/kit_ci.so, APP: KIT_CI
1980-012-14:03:20.25722 ES Startup: KIT_CI loaded and created
1980-012-14:03:20.25775 ES Startup: Loading file: /cf/ds.so, APP: DS
[... 88 lines cut ...]
EVS Port1 42/1/CFE_SB 25: Pipe Overflow,MsgId 0x808,pipe KIT_TO_PKT_PIPE,sender CS
EVS Port1 42/1/CS 127: OS Text Segment disabled due to platform
EVS Port1 42/1/CFE_SB 25: Pipe Overflow,MsgId 0x808,pipe KIT_TO_PKT_PIPE,sender CS
EVS Port1 42/1/CS 1: CS Initialized. Version 2.4.0.0
EVS Port1 42/1/CFE_SB 25: Pipe Overflow,MsgId 0x808,pipe KIT_TO_PKT_PIPE,sender CS
EVS Port1 42/1/ISIM 4: Successfully configured 10 initialization file attributes
EVS Port1 42/1/ISIM 123: ISIM table loaded.
EVS Port1 42/1/ISIM 25: Successfully replaced table 0 using file /cf/isim_tbl.json
EVS Port1 42/1/ISIM 100: ISIM App Initialized. Version 2.1.0
EVS Port1 42/1/HS 54: ExeCount verify results: good = 6, bad = 0, unused = 26
EVS Port1 42/1/HS 50: AppMon verify results: good = 1, bad = 0, unused = 31
EVS Port1 42/1/HS 52: EventMon verify results: good = 1, bad = 0, unused = 15
EVS Port1 42/1/HS 56: MsgActs verify results: good = 0, bad = 0, unused = 8
EVS Port1 42/1/HS 1: HS Initialized.  Version 2.3.0.0
EVS Port1 42/1/KIT_SCH 25: Successfully replaced table 1 using file /cf/osk_sch_sch_tbl.json
EVS Port1 42/1/KIT_SCH 101: KIT_SCH Initialized. Version 2.0.0
1980-012-14:03:21.17885 ES Startup: CFE_ES_Main entering APPS_INIT state
1980-012-14:03:21.17888 ES Startup: CFE_ES_Main entering OPERATIONAL state
EVS Port1 42/1/CF 73: SemGetId Err:Chan 0 downlink PDUs cannot be throttled.0xFFFFFFEF
EVS Port1 42/1/CFE_TIME 21: Stop FLYWHEEL
EVS Port1 42/1/KIT_TO 306: Telemetry output enabled for IP 127.0.0.1
EVS Port1 42/1/KIT_SCH 406: Multiple slots processed: slot = 0, count = 2
EVS Port1 42/1/SC 73: RTS Number 001 Started
EVS Port1 42/1/SC 86: RTS 001 Execution Completed
EVS Port1 42/1/KIT_SCH 404: Major Frame Sync too noisy (Slot 1). Disabling synchronization.

dmccomas, 2021-07-27:
> Hi, David!
>
> I installed today a new version of OSK and also fixed the reason for the
> messages ... Pipe Overflow, MsgId 0x808, pipe KIT_TO_PKT_PIPE ...
>
> And all I just need:
>
> As root, configure system queues' depth:
> Add the following lines to the end of the file /etc/sysctl.conf:
>
> # Increase message queues' depth
> fs.mqueue.msg_max = 500
>
> And also as root, configure default priority on a user.
> Add the following line to the file /etc/security/limits.conf:
>
> <user-name> - rtprio 100
>
> Example:
> #* soft core 0
> #root hard core 100000
> user - rtprio 100
> #* hard rss 10000
> #ftp hard nproc 0
> #ftp - chroot /ftp
>
> Thank you for your work!

Thank you for the inputs!!  I've been trying different startup sequences,
app priorities, and software bus queue depths to solve this issue.  My
thought process has been within the context of the cFS and not Linux.
Hopefully I can arrive at a consistent installation solution.

Thanks - Dave

APathak666, 2022-02-04:
> Oh, I seem to have solved my trouble!
>
> I just added the osk_42_lib to the list of applications for the cFS FSW Edu tab in source file ~/OpenSatKit-master/cfs/osk_defs/targets.cmake. Thus, in order [...]

I am facing the same problem (Major Frame Sync too noisy (Slot 1). Disabling synchronization.). I checked my "targets.cmake" file in osk_defs, and "osk_42_lib" is already added to the list of applications. However, I continue to get this error. Any idea what could be causing this?