OpenSatKit/OpenSatKit #75: OpenSatKit ERRNO::ENOENT : No such file or directory

hgarcia101685, 2021-08-11:
When trying to click on any of the buttons to open any of the guides or resources, I get the following error(screenshot provided)
ERRNO::ENOENT : No such file or directory

[image]

hgarcia101685, 2021-08-11:
Just solved my own issue, but will leave here in case it helps anyone else.  The evince(pdf display) package was missing on my ubuntu container, so running this fixed it for me:
`apt-get install evince`

closed 2021-08-11 by hgarcia101685

dmccomas, 2021-08-13:
This is dated 2015, but it states evince is the default ubuntu pdf viewer
https://help.ubuntu.com/community/Evince

I could add evince installation to the install script if needed.
Which version of Ubuntu are you using?

Thanks - Dave

hgarcia101685, 2021-08-14:
root@2972273c7169:~/OpenSatKit-master/cosmos/outputs/logs# cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"