OpenSatKit/OpenSatKit #83: how to start the whole cFS application

ryuxin, 2022-02-23:
I have successfully built the cFS source code and generated the executable binary.
After I run the binary `build/exe/cfsat/core-cfsat`, it seems the program is stuck or waiting for someting.
this is the output
```
[root@localhost cfsat]# ./core-cfsat 
CFE_PSP: Default Reset Type = PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cfsat
CFE_PSP: Starting the cFE with a POWER ON reset.
Policy 1: available, min-max: 1-99
Policy 2: available, min-max: 1-99
Selected policy 2 for RT tasks, root task = 99
Maximum user msg queue depth = 10
OSAL: Registering VolumeTable map for ./cf on /cf
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2032-053-01:50:02.71654 POWER ON RESET due to Power Cycle (Power Cycle).
2032-053-01:50:02.71655 ES Startup: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
2032-053-01:50:02.71659 ES Startup: CFE_ES_Main entering CORE_STARTUP state
2032-053-01:50:02.71660 ES Startup: Starting Object Creation calls.
2032-053-01:50:02.71660 ES Startup: Calling CFE_ES_CDSEarlyInit
2032-053-01:50:02.71664 ES Startup: Calling CFE_EVS_EarlyInit
2032-053-01:50:02.71665 Event Log cleared following power-on reset
2032-053-01:50:02.71666 ES Startup: Calling CFE_SB_EarlyInit
2032-053-01:50:02.71670 SB internal message format: CCSDS Space Packet Protocol version 1
2032-053-01:50:02.71671 ES Startup: Calling CFE_TIME_EarlyInit
1980-012-14:03:20.00000 ES Startup: Calling CFE_TBL_EarlyInit
1980-012-14:03:20.00007 ES Startup: Calling CFE_FS_EarlyInit
1980-012-14:03:20.00012 ES Startup: Core App: CFE_EVS created. App ID: 0
EVS Port1 42/1/CFE_EVS 1: cFE EVS Initialized. cFE Version 6.7.1.0
EVS Port1 42/1/CFE_EVS 14: No subscribers for MsgId 0x808,sender CFE_EVS
1980-012-14:03:20.05018 ES Startup: Core App: CFE_SB created. App ID: 1
1980-012-14:03:20.05020 SB:Registered 4 events for filtering
EVS Port1 42/1/CFE_SB 1: cFE SB Initialized
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_SB
1980-012-14:03:20.10022 ES Startup: Core App: CFE_ES created. App ID: 2
EVS Port1 42/1/CFE_ES 1: cFE ES Initialized
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/1/CFE_ES 2: Versions:cFE 6.7.1.0, OSAL 5.0.1.0, PSP 1.4.0.0, chksm 49216
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/1/CFE_ES 91: Mission v3.2-dirty.osk
EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 42/1/CFE_ES 92: Build 202202220945 root@localhost.localdomain
1980-012-14:03:20.15026 ES Startup: Core App: CFE_TIME created. App ID: 3
EVS Port1 42/1/CFE_TIME 1: cFE TIME Initialized
1980-012-14:03:20.20030 ES Startup: Core App: CFE_TBL created. App ID: 4
EVS Port1 42/1/CFE_TBL 1: cFE TBL Initialized.  cFE Version 6.7.1.0
1980-012-14:03:20.25032 ES Startup: Finished ES CreateObject table entries.
1980-012-14:03:20.25033 ES Startup: CFE_ES_Main entering CORE_READY state
1980-012-14:03:20.25034 ES Startup: Opened ES App Startup file: /cf/cfe_es_startup.scr
1980-012-14:03:20.25053 ES Startup: Loading shared library: /cf/cfs_lib.so
CFS Lib Initialized.  Version 2.2.0.01980-012-14:03:20.25080 ES Startup: Loading shared library: /cf/osk_c_fw.so
OSK C Application Framework Library Initialized. Version 2.0.0
1980-012-14:03:20.25105 ES Startup: Loading file: /cf/kit_to.so, APP: KIT_TO
1980-012-14:03:20.25128 ES Startup: KIT_TO loaded and created
1980-012-14:03:20.25151 ES Startup: Loading file: /cf/kit_ci.so, APP: KIT_CI
1980-012-14:03:20.25162 ES Startup: KIT_CI loaded and created
EVS Port1 42/1/KIT_CI 100: KIT_CI Initialized. Version 1.1.0
EVS Port1 42/1/KIT_TO 312: Removed 0 table packet entries
EVS Port1 42/1/KIT_TO 302: Successfully loaded new table with 37 packets
EVS Port1 42/1/KIT_TO 25: Successfully replaced table 0 using file /cf/osk_to_pkt_tbl.json
EVS Port1 42/1/KIT_TO 100: KIT_TO Initialized. Version 2.1.0
1980-012-14:03:20.30164 ES Startup: CFE_ES_Main entering APPS_INIT state
1980-012-14:03:20.30165 ES Startup: CFE_ES_Main entering OPERATIONAL state
EVS Port1 42/1/CFE_TIME 21: Stop FLYWHEEL
```

I enable the following cfs apps
```
SET(TGT1_APPLIST cfs_lib osk_c_fw kit_ci kit_sch kit_to osk_c_tutor)
SET(TGT1_FILELIST cfe_es_startup.scr osk_to_pkt_tbl.json osk_sch_msg_tbl.json osk_sch_sch_tbl.json osk_c_demo_ini.json osk_c_demo_tbl.json)
```

What should I do next to continue to run the cFS applications?

Thanks

dmccomas, 2022-02-24:
To use the cFS apps you can build the system using the preconfigured
targets in osk_defs.  The 'simsat' target uses most of the NASA cFS apps.

If you want to use COSMO with the cFS built using osk_defs then the 'Quick
Start Guide' in the docs directory is a good place to start.

ryuxin, 2022-02-25:
@dmccomas Thanks for your response.
I want to run cfs and 42 together within a closed loop.
Is `cosmos` necessary?
I do not have GUI environment, how can I run `cosmos` without GUI?

ryuxin, 2022-03-01:
@dmccomas I have built the `simsat` target and 42 simulator, and they seem to run normally.
I think `cosmos` or a ground-system is necessary to rum them together as `cosmos` issues commands to cFE.
However, I do not have GUI environment to run the complete `cosmos`.
Is there any way to run `cosmos` without GUI, such as directly running its ruby scripts?
Or any other methods to replace `cosmos`?

Thanks

dmccomas, 2022-03-01:
COSMOS is used as OSK's main application and assumes the platform supports
a GUI.  The COSMOS Command and Telemetry server is the single tool you must
run because it reads the cmd & tlm definition files and establishes an
interface to the cFS. I don't know whether it can be run without its GUI
and that's a question for the Ball Aerospace
https://github.com/BallAerospace/COSMOS . If it can be run headless then
you may have options other than ruby. I think they have an external JSON
interface and a python binding.

You may be able run the cFS python mini ground station without a GUI
https://github.com/nasa/cFS-GroundSystem/tree/b0faa754a62479dc9e97ffc7d389153ce9dd9e68
 The challenge will be that you will need to define the command and
telmeetry packets if they're not included with the tool.

ryuxin, 2022-03-03:
@dmccomas Thank you so much for your help.
I have figured out to install and run `cosmos` CmdTlmServer without GUI.
So I am able to run CmdTlmServer, cfs simsat target and 42 separately.
Then how can I run them together in a closed loop? I imagine I need to issue some commands via `cosmos`, but I do not know how to do that.

Could you provide some tutorial or guidance on this?

BTW, I list the output from CmdTlmServer
```
ruby tools/CmdTlmServer --no-gui 

2022/03/03 16:16:09.294  INFO: COSMOS Version: 4.4.0
2022/03/03 16:16:09.295  INFO: Created PATH LOGS /root/asid-isolation/cFS/OpenSatKit/cosmos/outputs/logs
2022/03/03 16:16:09.296  INFO: Created PATH SAVED_CONFIG /root/asid-isolation/cFS/OpenSatKit/cosmos/outputs/saved_config
2022/03/03 16:16:09.296  INFO: Created PATH HANDBOOKS /root/asid-isolation/cFS/OpenSatKit/cosmos/outputs/handbooks
2022/03/03 16:16:09.296  INFO: Created PATH PROCEDURES /root/asid-isolation/cFS/OpenSatKit/cosmos/cfs_kit/tutorials
2022/03/03 16:16:09.296  INFO: Created PATH PROCEDURES /root/asid-isolation/cFS/OpenSatKit/cosmos/config/targets/CFSAT/procedures
2022/03/03 16:16:09.297  INFO: Processing target definition in file '/root/asid-isolation/cFS/OpenSatKit/cosmos/config/targets/CFDP/target.txt'
2022/03/03 16:16:09.298  INFO: Processing target definition in file '/root/asid-isolation/cFS/OpenSatKit/cosmos/config/targets/CFDP_TEST/target.txt'
2022/03/03 16:16:09.300  INFO: Processing target definition in file '/root/asid-isolation/cFS/OpenSatKit/cosmos/config/targets/COSMOS/target.txt'
2022/03/03 16:16:09.304  INFO: Processing target definition in file '/root/asid-isolation/cFS/OpenSatKit/cosmos/config/targets/PDU/target.txt'
2022/03/03 16:16:09.308  INFO: Processing CmdTlmServer configuration in file: /root/asid-isolation/cFS/OpenSatKit/cosmos/config/tools/cmd_tlm_server/cmd_tlm_server.txt
INTERFACE LOCAL_CFS_INT udp_cs_interface.rb 127.0.0.1 1234 1235 nil nil 128 nil nil
2022/03/03 16:16:09.331  INFO: TFTP Task Initializing
2022/03/03 16:16:09.359  INFO: Marshal file does not exist: /root/asid-isolation/cFS/OpenSatKit/cosmos/outputs/tmp/marshal_ffcaba7b8986e971a966b6b6b6b6faf1.bin
2022/03/03 16:16:09.380  WARN: CF QUICK_STAT CCSDS_STREAMID redefined.
2022/03/03 16:16:09.380  WARN: CF QUICK_STAT CCSDS_SEQUENCE redefined.
2022/03/03 16:16:09.380  WARN: CF QUICK_STAT CCSDS_LENGTH redefined.
2022/03/03 16:16:09.381  WARN: CF QUICK_STAT CCSDS_CHECKSUM redefined.
2022/03/03 16:16:09.381  WARN: CF QUICK_STAT CCSDS_FUNCCODE redefined.
2022/03/03 16:16:09.570  WARN: FM WRITE_DIR_TO_FILE SPARE redefined.
2022/03/03 16:16:09.571  WARN: FM SEND_DIR_PKT SPARE redefined.
2022/03/03 16:16:09.611  WARN: Bit definition overlap at bit offset 144 for packet HS HK_TLM_PKT items SPARE_BIT_7 and STATE_BITS
2022/03/03 16:16:09.623  WARN: I42 SENSOR_DATA_PKT WHLH_1 redefined.
2022/03/03 16:16:09.623  WARN: I42 SENSOR_DATA_PKT WHLH_1 redefined.
2022/03/03 16:16:09.623  WARN: I42 SENSOR_DATA_PKT WHLH_1 redefined.
2022/03/03 16:16:09.661  WARN: KIT_TO HK_TLM_PKT SPARE_BYTE_1 redefined.
2022/03/03 16:16:09.672  WARN: Bit definition overlap at bit offset 128 for packet LC HK_TLM_PKT items WP_3_RESULT and WP_RESULTS
2022/03/03 16:16:09.738  WARN: Bit definition overlap at bit offset 576 for packet SC HK_TLM_PKT items RTS_8_EXE and RTS_W1_EXE_STATUS
2022/03/03 16:16:09.738  WARN: Bit definition overlap at bit offset 592 for packet SC HK_TLM_PKT items RTS_8_DIS and RTS_W1_DIS_STATUS
2022/03/03 16:16:10.969  INFO: Creating thread for router PREIDENTIFIED_ROUTER
2022/03/03 16:16:10.969  INFO: Creating thread for router PREIDENTIFIED_CMD_ROUTER
2022/03/03 16:16:10.969  INFO: Creating thread for interface COSMOSINT
2022/03/03 16:16:10.970  INFO: Creating thread for interface CFDP_INT
2022/03/03 16:16:10.970  INFO: Creating thread for interface LOCAL_CFS_INT
2022/03/03 16:16:10.970  INFO: Starting packet reading for PREIDENTIFIED_ROUTER
2022/03/03 16:16:10.970  INFO: Connecting to PREIDENTIFIED_ROUTER...
2022/03/03 16:16:10.970  INFO: PREIDENTIFIED_ROUTER Connection Success
2022/03/03 16:16:10.970  INFO: Starting connection maintenance for PREIDENTIFIED_CMD_ROUTER
2022/03/03 16:16:10.970  INFO: Connecting to PREIDENTIFIED_CMD_ROUTER...
2022/03/03 16:16:10.970  INFO: PREIDENTIFIED_CMD_ROUTER Connection Success
2022/03/03 16:16:10.970  INFO: Starting packet reading for COSMOSINT
2022/03/03 16:16:10.971  INFO: Connecting to COSMOSINT...
2022/03/03 16:16:10.971  INFO: COSMOSINT Connection Success
2022/03/03 16:16:10.971  INFO: TFTP Task Sleeping for 5 seconds after initial call
2022/03/03 16:16:10.971  INFO: Starting packet reading for CFDP_INT
2022/03/03 16:16:10.971  INFO: Starting packet reading for LOCAL_CFS_INT
2022/03/03 16:16:10.971  INFO: Connecting to LOCAL_CFS_INT...
2022/03/03 16:16:10.972  INFO: CFDP Engine has started.
2022/03/03 16:16:10.972  INFO: LOCAL_CFS_INT Connection Success
2022/03/03 16:16:14.974  INFO: Log File Opened : /root/asid-isolation/cFS/OpenSatKit/cosmos/outputs/logs/2022_03_03_16_16_14_tlm.bin
2022/03/03 16:16:15.976  INFO: TFTP Task Running

```

dmccomas, 2022-05-20:
Apologies for the delay.  
Your screen shot shows that the COSMOS command and telemetry server is running.  The next step is to start the cFS. You can do the by clicking  "Start cFS" on the cFS Education tab.  A new terminal window will open and you'll be prompted for your password to run the cFS. The 'OSK Quick Start Guide' under 'OSK Docs & Videos' goes into more detail. 

[image]