OpenSatKit/OpenSatKit #85: Some problems with the instalation

octavio-luna, 2022-03-18:
Hello everyone! I´m trying to install OpenSatKiton on a VM with ubuntu 16.04 using _bash <(\wget -qO- https://raw.githubusercontent.com/OpenSatKit/OpenSatKit/vendor/install.sh)_

When I get into OpenSatKit-master/cosmos and run ruby Launcher, I get the following error:

> Traceback (most recent call last):
	3: from Launcher:13:in `<main>'
	2: from /home/octavio/OpenSatKit-master/OpenSatKit/cosmos/tools/tool_launch.rb:14:in `tool_launch'
	1: from /usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:59:in `require'
/usr/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:59:in `require': cannot load such file -- bundler/setup (LoadError)

Does anyone knows where might the problem be? 

Really thanks you in advantage!

dmccomas, 2022-03-18:
You can try the following  two commands.

   gem install bundler  (or sudo gem install bundler if privileges are
required)
   bundle install

Hopefully this doesn't indicate a broader issue of the ruby environment not
properly installing. If possible I would recommend upgrading to Ubuntu
18.04 or 20.4. Hope it works.

octavio-luna, 2022-03-21:
Hi @dmccomas thanks for the response! I´ve tried it. I´ve also uograded my ubuntu version to 20.04, but yhr installation still not works. I´m getting this issue when trying to run bundle install (since ruby Launcher requires to run bundle install to install missing gems):

> An error occurred while installing json (2.6.1), and Bundler cannot
> continue.
> Make sure that `gem install json -v '2.6.1' --source 'https://rubygems.org/'`
> succeeds before bundling.

> In Gemfile:
>   cosmos was resolved to 4.5.2, which depends on
>     json
> 
> 

And when I try to run gem install json I get the next issue:

> ERROR:  While executing gem ... (Gem::FilePermissionError)
>     You don't have write permissions for the /var/lib/gems/2.7.0 directory.

octavio-luna, 2022-03-22:
UPDATE:
I´ve managed to get json gem installed by installing it via rbenv. After that, I´ve had some issues with termios 0.9.6 installation (i´ve googled it and it seems to be a rellatively common issue with that particular version) so I´ve changed the dependencies to termios 1.1.0 in cosmos/Launcher.rb since that´s the last stable version. But now I´m getting a new error and I can´t figure out what may I do:

> /home/octavio/.rbenv/versions/3.0.2/lib/ruby/gems/3.0.0/gems/cosmos-0.0.6/lib/cosmos/service.rb:9: warning: already initialized constant Middleware::Runner::EMPTY_MIDDLEWARE
> /home/octavio/.rbenv/versions/3.0.2/lib/ruby/gems/3.0.0/gems/middleware-0.1.0/lib/middleware/runner.rb:7: warning: previous definition of EMPTY_MIDDLEWARE was here
> Launcher:14:in `require': cannot load such file -- cosmos/tools/launcher/launcher (LoadError)
> 	from Launcher:14:in `block in <main>'
> 	from /home/octavio/OpenSatKit-master/cosmos/tools/tool_launch.rb:16:in `tool_launch'
> 	from Launcher:13:in `<main>'

It seems like there´s no such file like `cosmos/tools/launcher/launcher`. Anyone has any idea what should I do?

dmccomas, 2022-04-14:
COSMOS issue #920 may be related to some of the issues in this thread. issue: https://github.com/BallAerospace/COSMOS/issues/920
v3.4 will fixes this Gemfile problem but I'lll leave this issue open.

To fix OSK edit OpenSatKit/cosmos/GemfIle
and change from

gem 'ruby-termios', '~> 0.9'

to 

gem 'ruby-termios', '>= 0.9'

Hopefully this fixes the issue.

dmccomas, 2022-05-24:
I think the problem is related to OSK's Gemfile.lock being in sync with the COSMOS release. You may be able to do 'bundle update cosmos' with your current installation. If that doesn't work then you should be able to install OpenSatKit version 4.0. I just released v4.0 today and tested the installation on Ubuntu 18.04 and it worked without the need to do a "bundle install' after the installation. I'll test Ubuntu 20.04 soon.