OpenSatKit/OpenSatKit #86: CCSDS v2 Packets

space-individual, 2022-04-08:
Hi,
I am trying to change the CCSDS packets to support version 2.  By default it looks like the packets are using version 1.  Is there a way to easily change this in COSMOS?  I know CFS supports the version 2 packets.
Any help would be greatly appreciated!

dmccomas, 2022-04-09:
Yes it should be fairly straightforward to make the changes and this is
something that's on the OSK roadmap. I wasn't sure how big the demand was
so it was in the backlog and there's the issue of cFS versions. OSK uses
cFS Framework Aquila because that's the latest version that all of the NASA
apps have been released to.  CCSDS version 2 support was added in Aquila
but I don't know the extent to which it was tested.  I'd like to transition
to Caelum that has had extensive testing but then I'd have the app issue.
 OSK is being refactored and hopefully I can announce/release it in about a
month.

OSK's _cmd.txt files use embedded ruby to instantiate the CCSDS headers
defined in cosmos/lib/ccsds.rb so if you make your changes to ccsds.rb then
every packet will be changed. This line in each command definition
initiates the command header. The last two integers are the command
function code and the number of command payload bytes (excludes header)  so
this example has a function code of 4 and 128 payload bytes.
<%= ***@***.***_PREFIX_STR, @CMD_MID_STR, 4, 128) %>

I've started an OSK COSMOS guide which goes into more detail

https://github.com/OpenSatKit/OpenSatKit/blob/master/docs/OSK-COSMOS-Guide.pdf

space-individual, 2022-04-11:
Thanks @dmccomas.  This is off topic, but your videos on OpenSatKit have been some of the best information that I can find regarding the Core Flight System, and have helped a ton with the project I am currently working on!

closed 2022-04-11 by space-individual

dmccomas, 2022-04-13:
Thanks for the feedback and glad it was helpful.  I've partnered with Space
Steps https://spacesteps.com/ and we hope to get back to video production
in the next few months.  - Dave