OpenSatKit/OpenSatKit #91: Cannot load Ruby file

AIoTvn, 2022-05-10:
I've successfully installed bundle with " bundle install ". But when I run ruby launcher i faced with this issue. Pls help me, thank you very much !
" Bundle complete! 2 Gemfile dependencies, 37 gems now installed.
Use `bundle info [gemname]` to see where a bundled gem is installed.
dnm@mm:~/OpenSatKit-master/cosmos$ ruby Launcher
Traceback (most recent call last):
3: from Launcher:13:in `<main>'
2: from /home/dnm/OpenSatKit-master/cosmos/tools/tool_launch.rb:16:in `tool_launch'
1: from Launcher:14:in `block in <main>'
Launcher:14:in `require': cannot load such file -- cosmos/tools/launcher/launcher (LoadError)
"

koben98, 2022-05-10:
Glad to know I'm not the only one facing this issue. Hope to figure out why soon

koben98, 2022-05-10:
The issue might relate to [Problem #85](https://github.com/OpenSatKit/OpenSatKit/issues/85). Hope that helps

DBprod, 2022-05-11:
Hello, I have this issue as well, I am on Ubuntu 22.04 VMware VM.

dmccomas, 2022-05-12:
I'm wondering whether these issues are related to a recent change I made to
the cosmos/Gemfile based on the following cosmos ticket:
https://github.com/BallAerospace/COSMOS/issues/920

I changed

gem 'ruby-termios', '~> 0.9'

to
    gem 'ruby-termios', '>= 0.9'

I'll do some testing with the two configurations...

koben98, 2022-05-12:
I tried switching it to '> 0.9', same error on Ubuntu 18.04, 20.04 and 22.04. Both VMs and boot directly on a laptop if there's any difference.
Most recent attempt did not include ruby, I installed it by the default sudo apt install ruby as well as rbenv with 2.4.0-dev, still the same error

dmccomas, 2022-05-15:
I think the problem is related to OSK's Gemfile.lock being in sync with the
COSMOS release. You may be able to do  'bundle update cosmos' with your
current installation.

If that doesn't work then you should be able to install OpenSatKit version
4.0. I just released v4.0 today and tested the installation on Ubuntu 18.04
and it worked without the need to do a "bundle install' after the
installation. I'll test Ubuntu 20.04 soon.

- Dave

AIoTvn, 2022-05-17:
I solved by run terminal from first steps and it's worked in all of the devices i've tried.

minhtn1, 2022-05-22:
> I think the problem is related to OSK's Gemfile.lock being in sync with the COSMOS release. You may be able to do 'bundle update cosmos' with your current installation. If that doesn't work then you [...]

Have u tested on ubuntu 20.04 :(( i haved same issues on ubuntu 20.04

dmccomas, 2022-07-12:
There may be multiple issues being discussed in this issues but I wanted to loop back to the original entry that resulted in the following error when 'ruby Launcher' is entered:
Launcher:14:in `require': cannot load such file -- cosmos/launcherlauncher (LoadError)

This may indicate that ruby wasn't properly installed during the the COSMOS installation. It's very easy to make a typo while answering some of the prompts during the installation. 
- Review the installation  messages. It would be worth adding a note to the installation script to save the output prior to closing the window. 
- Verify ruby is installed by typing 'ruby --version'

nguyentuanngoc21, 2022-07-22:
> There may be multiple issues being discussed in this issues but I wanted to loop back to the original entry that resulted in the following error when 'ruby Launcher' is entered: Launcher:14:in [...]

my version is ruby 3.0.2p107 (2021-07-07 revision 0db68f0233), and I think it is up-to-date version, but I still meet this error

AIoTvn, 2022-08-17:
May I ask you a question that what type of C that cFS use ?? I’ve found but
i still not have any information about that. Thank you very much .

dmccomas, 2022-08-18:
The cFS is built using the -std=C99 flag  Here's a C99 reference: https://en.m.wikipedia.org/wiki/C99