OpenSatKit/OpenSatKit #92: Can't run Helloworld app

AIoTvn, 2022-05-17:
At step 4 of Create Helloworld app, I cant Stop cFS cmd tlm Server. On terminal say " Sorry, try again ". I followed #issue68 but it didn't work. What can I do :(( . I'm using ubuntu 20.04, cosmos 4.5.2. What can I do ?? Thank you very much !

AIoTvn, 2022-05-17:
i received this text output : 
" 2022/05/17 15:41:39.517  FATAL: Fatal Exception! Exiting...
2022/05/17 15:41:39.518  FATAL: RuntimeError : Error binding to port 7777.
Either another application is using this port
or the operating system is being slow cleaning up.
Make sure all sockets/streams are closed in all applications,
wait 1 minute and try again.
/home/dnm/.rbenv/versions/2.5.8/lib/ruby/gems/2.5.0/gems/cosmos-4.5.2/lib/cosmos/io/json_drb.rb:169:in `rescue in block in start_service'
/home/dnm/.rbenv/versions/2.5.8/lib/ruby/gems/2.5.0/gems/cosmos-4.5.2/lib/cosmos/io/json_drb.rb:132:in `block in start_service'
"

dmccomas, 2022-05-24:
You can manually stop the COSMOS Command and Telemetry Server by closing the window. That's all the button script tries to do. I'm wondering whether it's a permission issue with why it didn't work.