OpenSatKit/OpenSatKit #97: Error when build cFS with new ' cFE App Dev Guide template ' applications

AIoTvn, 2022-10-10:
Hello, I've created a new application with ' cFE App Dev Guide template '. But when I build new cFS i received this message :
[image]
Can you help me pls @dmccomas, many thanks.

dmccomas, 2022-10-11:
I'm traveling through 10/21 and do not have access to a computer where I can try to replicate the issue. We should still be able to resolve it. 

The 'install' process is copying files object and tables files into the build/exe/cfsat/cf/ directory. This error indicates the table file didn't get built. .tbl files are built as follows:
1. The table source code is compiled into an ELF object file
2. The elf2cfetbl tools is used to create a .tbl file from the ELF file

Since a code generation tool is being used let's confirm the table code even got generated correctly. Do you see table source files in cfs/apps/app_dev_example1/fsw/tables/ ?

If not then there's a template problem and I'll have to fix it. 

if the source exists then did the source get compiled? I looked at the CMakeLists.txt template code in github and it looks correct. I think a good approach here would be to start with a fresh cmake build environment in case it didn't pick of the new tables. Issue the following:
make distclean

followed by:
make prep
make install

Look at the messages output during the build process for your app. They should be similar to messages for another app that has tables. For example I see Memory Dwell (MD) in your log.  You should be able to see where MD tables gets compiled and also when elf2cfetbl gets run.  If the install still fails please let me know which other step did not occur.

AIoTvn, 2022-10-13:
Yes t checked and table code has generated successfully. When run ' make install ' faced that warning, i don't know if it was the reason but i think it wasn't. 
[image]
And when i try with ' cFE Sample App ", it build successfully. But when cFS run i faced this error : 
[image]
At the moment i just want to learn to modify a template to get an application that can communicate with COSMOS.  Can you give me some advice. Thank you so much!!

AIoTvn, 2022-10-13:
> And when i try with ' cFE Sample App ", it build successfully. But when cFS run i faced this error :

I mean ' cFE Sample App ' template.

highter9, 2023-06-09:
I got the same error and fixed it. Maybe your table name is too long. I remember there is a limitation of 20 characters somewhere.