OpenSatKit/OpenSatKit #99: Installation error

PratikAher20, 2022-12-13:
I tried copying and then pasting the link into the command terminal but the execution got stuck showing "Redirecting output to wget-log". Please help to rectify this issue.

dmccomas, 2022-12-14:
I'm not familiar with this issue but I did find this post and they
suggested adding "-q' as the first wget parameter to resolve the issue.
linux - wget now automatically redirecting output to a log file, how to
return back to putting output below - Stack Overflow
<https://stackoverflow.com/questions/52214260/wget-now-automatically-redirecting-output-to-a-log-file-how-to-return-back-to-p#:~:text=Redirecting%20output%20to%20%E2%80%98wget-log.4%E2%80%99.%20This%20feature%20was%20introduced,A%20reliable%20workaround%20is%20to%20add%20-o%20%2Fdev%2Fnull.>

- Dave

PratikAher20, 2022-12-15:
For new users of Ubuntu on VM if you getting similar error as above do follow this https://florahuo.medium.com/edit-etc-hosts-file-in-ubuntu-command-line-using-nano-ide-44c3a78e23ca

closed 2022-12-15 by PratikAher20