OpenSatKit/OpenSatKit #100: Connection refused when I add new SC

sensationHJ, 2022-12-20:
Hi !
I am trying to add new SC to 42 in OSK.
I edit `OpenSatKit-master/42/OSK/Inp_Sim.txt` and run, but I get this messages
```sh
EVS Port1 42/2/CFE_ES 3: No-op command. Versions:cFE 6.7.1.0, OSAL 5.0.1.0, PSP 1.4.0.0
EVS Port1 42/2/KIT_SCH 406: Multiple slots processed: slot = 1, count = 2
EVS Port1 42/2/I42 122: Error connecting client socket: Connection refused
EVS Port1 42/2/KIT_SCH 406: Multiple slots processed: slot = 1, count = 2
EVS Port1 42/2/KIT_SCH 406: Multiple slots processed: slot = 1, count = 2
EVS Port1 42/2/KIT_SCH 404: Major Frame Sync too noisy (Slot 1). Disabling synchronization.
```

this is `Inp_Sim.txt` i edited and `Inp_IPC.txt`

`Inp_Sim.txt`
```
<<<<<<<<<<<<<<<<<  42: The Mostly Harmless Simulator  >>>>>>>>>>>>>>>>>
************************** Simulation Control **************************
FAST                            !  Time Mode (FAST, REAL, or EXTERNAL)
2000.0   0.2                    !  Sim Duration, Step Size [sec]
1.0                             !  File Output Interval [sec]
TRUE                            !  Graphics Front End?
Inp_Cmd.txt                     !  Command Script File Name
**************************  Reference Orbits  **************************
1                               !  Number of Reference Orbits
TRUE   Orb_Cube.txt              !  Input file name for Orb 0
*****************************  Spacecraft  *****************************
1                               !  Number of Spacecraft
TRUE  0 SC_Cubesat6U.txt             !  Existence, RefOrb, Input file for SC 0
***************************** Environment  *****************************
03 21 2016                      !  Date (UTC) (Month, Day, Year)
12 00 00.00                     !  Time (UTC) (Hr,Min,Sec)
0.0                             !  Leap Seconds (sec)
USER_DEFINED                    !  Model Date Interpolation for Solar Flux and AP values?(TWOSIGMA_KP, NOMINAL or USER_DEFINED)
230.0                           !  If USER_DEFINED, enter desired F10.7 value
100.0                           !  If USER_DEFINED, enter desired AP value
IGRF                            !  Magfield (NONE,DIPOLE,IGRF)
8   8                           !  IGRF Degree and Order (<=10)
8   8                           !  Earth Gravity Model N and M (<=18)
2   0                           !  Mars Gravity Model N and M (<=18)
2   0                           !  Luna Gravity Model N and M (<=18)
FALSE   FALSE                   !  Aerodynamic Forces & Torques (Shadows)
FALSE                           !  Gravity Gradient Torques
FALSE   FALSE                   !  Solar Pressure Forces & Torques (Shadows)
FALSE                           !  Gravity Perturbation Forces
FALSE                           !  Passive Joint Forces & Torques
FALSE                           !  Thruster Plume Forces & Torques
FALSE                           !  RWA Imbalance Forces and Torques
FALSE                           !  Contact Forces and Torques
FALSE                           !  CFD Slosh Forces and Torques
FALSE                           !  Output Environmental Torques to Files
********************* Celestial Bodies of Interest *********************
MEAN                            !  Ephem Option (MEAN or DE430)
FALSE                           !  Mercury
FALSE                           !  Venus
TRUE                            !  Earth and Luna
FALSE                           !  Mars and its moons
FALSE                           !  Jupiter and its moons
FALSE                           !  Saturn and its moons
FALSE                           !  Uranus and its moons
FALSE                           !  Neptune and its moons
FALSE                           !  Pluto and its moons
FALSE                           !  Asteroids and Comets
***************** Lagrange Point Systems of Interest ******************
FALSE                           !  Earth-Moon
FALSE                           !  Sun-Earth
FALSE                           !  Sun-Jupiter
************************* Ground Stations ***************************
5                                            ! Number of Ground Stations
TRUE  EARTH  -77.0  37.0  "GSFC"             ! Exists, World, Lng, Lat, Label
TRUE  EARTH  -155.6 19.0  "South Point"      ! Exists, World, Lng, Lat, Label
TRUE  EARTH  115.4 -29.0  "Dongara"          ! Exists, World, Lng, Lat, Label
TRUE  EARTH  -71.0 -33.0  "Santiago"         ! Exists, World, Lng, Lat, Label
TRUE  LUNA   45.0  45.0   "Moon Base Alpha"  ! Exists, World, Lng, Lat, Label
```

`Inp_IPC.txt`
```
<<<<<<<<<<<<<<< 42: InterProcess Comm Configuration File >>>>>>>>>>>>>>>>
1                                       ! Number of Sockets
**********************************  IPC 0   *****************************
ACS                                     ! IPC Mode (OFF,TX,RX,TXRX,ACS,WRITEFILE,READFILE)
0                                       ! AC.ID for ACS mode
"State00.42"                            ! File name for WRITE or READ
SERVER                                  ! Socket Role (SERVER,CLIENT,GMSEC_CLIENT)
localhost     10001                     ! Server Host Name, Port 
TRUE                                    ! Allow Blocking (i.e. wait on RX)
FALSE                                   ! Echo to stdout
1                                       ! Number of TX prefixes
"SC[0].AC"                              ! Prefix 0
```

I just replace Orbit and SC txt from `OpenSatKit-master/42/Demo` folder.. is there other configs should I have to know..?

Any help or insight would be great...!

thanks!

dmccomas, 2022-12-21:
I think there may be a couple of issues

1.  It looks like the 42 simulator isn't running. The 42 simulator must be
started from the Mission FSW tab shown below. The easiest way to start 42
with the cFS is to use the "Start cFS with 42 Sim" button. It runs a script
that takes care of startup order.  I42 is a cFS app that manages the 42
interface. If 42 is running but the cFS isn't connected to it you can issue
an I42  "Connect 42" command.

2.  The second issue is that opensatkit runs 42 from the
OpenSatKit-master\42\OSK directory so you'll need to change files in that
directory.

[image: image.png]

Unfortunately the cFS-42 system documentation is on the "to do" list.
There's a little design  information in
OpenSatKit-master\cosmos\config\targets\SIMSAT\docs\Mission-Fsw-Apps-Adc.pdf

Let me know if this is unclear or if you have any other questions.

- Dave

sensationHJ, 2023-01-03:
Thank you for your response!

I checked the configuration folder used with 42 is OpenSatKit-master\42\OSK before from ruby scripts :)

`OpenSatKit-master/cosmos/lib/osk_system.rb`
```ruby
      def self.start_42(interactive=true, display_scr=false)

         started_42 = false
         if (not Osk::System.check_n_start_cfs("prompt",interactive))
            return started_42
         end
         if tlm("I42 HK_TLM_PKT CONNECTED") == "TRUE"
            message_box("42 not started. It's already running and connected.",Osk::MSG_BUTTON_CONT,false) if interactive
            return started_42
         end
         
         started_42 = true
         osk_42_dir = File.join(Osk::OSK_42_DIR,"OSK")
         include_gui = message_box("Would you like to run 42 with a GUI?", Osk::MSG_BUTTON_YES, Osk::MSG_BUTTON_NO, false)
         if (include_gui == Osk::MSG_BUTTON_YES)
            system("cp #{osk_42_dir}/Inp_Sim-gui.txt #{osk_42_dir}/Inp_Sim.txt")
         else
            system("cp #{osk_42_dir}/Inp_Sim-no_gui.txt #{osk_42_dir}/Inp_Sim.txt")
         end
         spawn("xfce4-terminal --title=\"42 Simulator\" --default-working-directory=\"#{Cosmos::USERPATH}/#{Osk::REL_DIR_42}\" --execute ./42 OSK")
         wait(4)
         Osk::flight.send_cmd("I42","CONNECT_42")
        
         display("SIMSAT SIM_42_SCREEN",1500,50) if display_scr
         
         return started_42
      
      end # start_42()
```

so I edit Orbit and Spacecraft part of `Inp_Sim-gui.txt`  

```**************************  Reference Orbits  **************************
1                               !  Number of Reference Orbits
TRUE   Orb_Cube.txt              !  Input file name for Orb 0
*****************************  Spacecraft  *****************************
1                               !  Number of Spacecraft
TRUE  0 SC_Cubesat6U.txt             !  Existence, RefOrb, Input file for SC 0
```

I made SC, Orbit configuration in 42 cloned from https://github.com/ericstoneking/42 based on Demo and StandAlone, RX, TX Folder.

and It worked well at that repo.

but when I brought that to 42 in OSK, it didnt work..

After many trial..  I found the reason

I think.. The reason was the version..!

SC_xxx.txt have little differences between [ericstoneking/42](https://github.com/ericstoneking/42) master with [OSK/42](https://github.com/OpenSatKit/OpenSatKit/tree/master/42)

So .. if someone want to edit OSK's 42 model.. do not start with ericstoneking/42 😂

I got order to find satellite simulator, so after finding simulator.. my start was ericstoneking/42 haha

closed 2023-01-03 by sensationHJ