OpenSatKit/OpenSatKit #101: ruby Launcher

vicjunejr, 2022-12-28:
Hello everyone! I am trying to install OpenSatKit in Ubuntu 18.04. After running the 'bundle install', which installed the missing gems successfully, trying to open the launcher, I received this error:

vicjunejr@vicjunejr-PC:~/OpenSatKit-master/cosmos$ ruby Launcher
Calling `DidYouMean::SPELL_CHECKERS.merge!(error_name => spell_checker)' has been deprecated. Please call `DidYouMean.correct_error(error_name, spell_checker)' instead.
Launcher:14:in `require': cannot load such file -- cosmos/tools/launcher/launcher (LoadError)
	from Launcher:14:in `block in <main>'
	from /home/vicjunejr/OpenSatKit-master/cosmos/tools/tool_launch.rb:16:in `tool_launch'
	from Launcher:13:in `<main>'

Does anyone knows how to resolve this?
A big thanks in advance!

dmccomas, 2022-12-28:
I've never seen that specific error but Issue 91 identifies some other ruby
Launcher issues. Cannot load Ruby file · Issue #91 · OpenSatKit/OpenSatKit
(github.com) <https://github.com/OpenSatKit/OpenSatKit/issues/91>

Can you try 'bundle update cosmos' to see whether that helps.

jjujju95, 2023-02-01:
I also had the same error as above, has anyone solved it?
my ubuntu version 20.04, ruby version also 2.7.0p0

clxyder, 2023-04-30:
I ran into the same issue earlier this year and was able to successfully run OSK, after some troubleshooting.

But I don't quite remember exactly what I did, but I believe I manually re-installed ruby (verify the correct version) and all of it's dependencies and then updated them.

This is what I get when I run `--version` for ruby, gem and bundle. Note, I am running this on WSL.

```bash
ruby --version
ruby 2.5.8p224 (2020-03-31 revision 67882) [x86_64-linux]

gem --version
2.7.6.2

bundle --version
Your RubyGems version (2.7.6.2) has a bug that prevents `required_ruby_version` from working for Bundler. Any scripts that use `gem install bundler` will break as soon as Bundler drops support for your Ruby version. Please upgrade RubyGems to avoid future breakage and silence this warning by running `gem update --system 3.2.3`
Bundler version 2.3.26
```

dmccomas, 2023-05-03:
Unfortunately OpenSatKit may be nearing its end-of-life since Ball isn't
supporting it any more.  I have not been successful with a clean install on
Ubuntu 20.04. I plan to refactor OpenSatKit to use OpenC3's COSMOS
https://openc3.com/ but I can't say when I'll have enough time.

I have a new cFS tool called cFS Basecamp,
https://github.com/cfs-tools/cfs-basecamp  Basecamp provides a bottom up
approach for learning the cFS. It has a simple Python GUI so easy
installation, but it's not close to a full featured ground system.  I plan
to make some cFS educational videos using examples from Basecamp. Basecamp
uses Electronic Data Sheets which are part of the cFS roadmap so hopefully
we'll converge in the future.

- Dave