OpenSatKit/OpenSatKit #104: Issue with installing and running cFS

katelynb-oss, 2023-03-02:
Unable to start flight software 

Hello,

I recently downloaded OpenSatKit on Ubuntu 20.04. 

I think I am having a similar issue to what is described in these posts: https://github.com/OpenSatKit/OpenSatKit/issues/44 and https://github.com/OpenSatKit/OpenSatKit/issues/58

When I click the green Start cFS button, a terminal window opens up but closes after I enter my password. The system time does not increment. When I select "Run Demo" on the cFE event service screen, a message box tells me the flight software is not running. When when I click "yes" to start the flight software, the terminal window just disappears after the enter my password. 

After reading this post (https://github.com/OpenSatKit/OpenSatKit/issues/58), it doesn't seem like cFS wasn't built during installation. The file ~/OpenSatKit-master/cfs/build/exe/cpu1/core-cpu1 does not exist for me. I only have  ~/OpenSatKit-master/cfs/build/exe in common with that file path. When I follow your steps instructed in  https://github.com/OpenSatKit/OpenSatKit/issues/44 and https://github.com/OpenSatKit/OpenSatKit/issues/58
by  going to  ~/OpenSatKit-master/cfs/  and entering: 

make prep
make install

it looks like it installed with no errors (I couldn't see any error flags in the text output). I attached the terminal window output when I send those two commands FYI.

However, this doesn't seem to solve the issue. The terminal window still closes after I enter my password and it doesn't seem as though the flight software is running. Do you have any ideas on what I should try next ?

Thanks, looking forward to hearing back
Katelyn

[install_error.txt](https://github.com/OpenSatKit/OpenSatKit/files/10875357/install_error.txt)

dmccomas, 2023-03-03:
Hi Katelyn,

Thanks for the thorough information, it's very helpful.  I think the
quickest way to get you started is to run the cFS in a second terminal. The
cFS does not have to be started from the COSMOS GUI. Then we can try to
debug the terminal window closing.

Two cFS images are built. One is called cfsat and the other is called
simsat. Both were successfully built based on your log. I recommend you
start with cfsat since it has less apps running and you can work through
some introductory material.

To run the cfsat image from a terminal window:
1. cd /home/ksball/OpenSatKit-master/cfs/build/exe/cfsat
2. sudo ./core-cfsat
3. This starts the cFS. You should see a bunch of messages in the
terminal window ending with CFS_ES entering an operational state. The
FLYWHEEL and slots missed messages can be ignored
[image: image.png]
4. From the main COSMOS GUI select "Enable Telemetry" from the "Configure
cFS System". This starts telemetry and you should see time start to update.
When the "Start cFS" button works this step is automatically performed by a
ruby script.

[image: image.png]

Let me know how it goes.

It's also good to know OSK installed on Ubuntu 20.04.  OSK may have a
limited shelf life since COSMOS isn't being maintained.

- Dave

katelynb-oss, 2023-03-07:
Hi Dave

Thanks for the information! So I went to start cfs in a separate terminal and I get the following error: 

./core-cfsat: 1: Syntax error: "(" unexpected 

I attached the error_2 text file which shows the error in the terminal window. 
[error_2.txt](https://github.com/OpenSatKit/OpenSatKit/files/10912911/error_2.txt)

I tried these steps in a second Ubuntu terminal after I already launched cosmos GUI, before I launched cosmos GUI and with the simsat image. 

I am not so familiar with running such programs in Linux, so I am not sure how common this error is, or if I am doing something fairly obvious wrong. 

Thanks for you continued support :) 

Katelyn

dmccomas, 2023-06-03:
Hi Katelyn,

I apologize for going silent. I've been struggling trying to balance my
time.  Were you able to get past this? The "(" error has me baffled.

I have a new cFS project called cFS Basecamp
https://github.com/cfs-tools/cfs-basecamp which is where I've been
spending a lot of time.  Depending on your needs this may be of some use.

- Dave