OpenSatKit/OpenSatKit #111: OSK_C_TUTOR - Undefined Method "fsw_name"

clxyder, 2023-05-02:
After building the application from the GUI, I then attempt to run the osk_c_tutor app, but I run into the following problem:

[image]

This happened for all the lessons, I hoped that by the last lesson it would have fixed itself.

Do I need to add something to `cfs/osk_defs/cfsat_cfe_es_startup.scr`?

clxyder, 2023-05-03:
I was able to make some progress by adding the following line to my `cfs/osk_defs/cfsat_cfe_es_startup.scr`

```
CFE_APP, /cf/osk_c_tutor.so,  OSK_C_DEMO_AppMain,  OSK_C_DEMO,    90,   16384, 0x0, 0;
```

But that would make the cFS crash immediately when booted up.

So I swapped `osk_c_tutor` to `osk_c_demo` in:

```
SET(TGT1_APPLIST cfs_lib osk_c_fw kit_ci kit_sch kit_to tftp fm md mm osk_c_tutor example)
```

Which would now allow for the cFS to boot up without crashing, but when I pressed the `Screen` button in the `CFSAT OSK_C_LESSON_SCR` window and the pressed the `NOOP` button to send a no-op command I got an error that said something along the lines of:

```
send_command is not defined for class nil:NilClass
```

dmccomas, 2023-05-03:
Carlos,

It's been a while since I've looked at osk_c_tutor. I should have time by
the end of the week to dig into it.

- Dave

clxyder, 2023-05-04:
Thank you @dmccomas I will keep playing around with it and report back.

clxyder, 2023-05-05:
If I press `Ops Example` from the `OSK_D_DEMO DEMO_OPS_SCREEN` window and then press `Start` the script progresses to line 131:

```
Osk::flight.send_cmd("OSK_C_DEMO", "NOOP")
```

Which is when I get the following error in the `Script Output` section:

```
2023/05/05 12:23:50.323 (osk_c_demo_ops_example.rb:131): ERROR: RuntimeError : Undefined application OSK_C_DEMO. Verify COSMOS target defined and target defined in ~/OpenSatKit-master/cosmos/config/targets/CFS_KIT/osk/cfs_kit.json
```

Is there anything specific I need to add to that json file?

clxyder, 2023-05-05:
Okay I was able to fix the issue.

We need to add `OSK_C_DEMO` to the list associated with the `app` key in `cosmos/config/targets/CFSAT/osk/cfsat.json`

```
"apps": [ "FM", "KIT_CI", "KIT_SCH", "KIT_TO", "MD", "MM", "TFTP", "OSK_C_DEMO"]
```

dmccomas, 2023-05-08:
Thanks for the good detective work! I'm reopening the issue and will fix the "apps": [ "FM", "KIT_CI", "KIT_SCH", "KIT_TO", "MD", "MM", "TFTP", "OSK_C_DEMO"].

There's additional cleanup I need to do since I removed the sandbox target. Prior to the sandbox target removal, OSK_C_DEMO was delivered as part of the sandbox and OSK_C_TUTOR was developed within the context of the CFSAT target. Now only CFSAT exists so this strategy should be reworked within the context of a single CFSAT target.

clxyder, 2023-05-10:
Would you like for me to open a PR? Or is there something I can help you with?

dmccomas, 2023-06-09:
Hi Carlos,

I reopened the issue hoping I would fix it and I appreciate the offer to
help.  It looks like I won't be able to maintain OpenSatKit for multiple
reasons.

I'm still committed to cFS education and I've created a new tool called cFS
Basecamp https://github.com/cfs-tools/cfs-basecamp and It will have coding
exercises. I'm creating new videos that will be posted in the following
channel:  OpenSatKit - YouTube
<https://www.youtube.com/channel/UC2wfvAIkrrgyC4ITwL3zokg>

Warm regards,
- Dave

clxyder, 2023-06-10:
Thanks for the response, I will leave the PR open for now. I will try to get a PR out, if not I will close this issue.